jilleb/mqb-pm #1: Not supported firmware versions

jilleb, 2018-02-03:
Currently, the following firmwares do not have an exlap service whitelist for Android Auto, which means there will be no data in the app.

VW 0245
VW 0343
VW 0613
VW 0751
VW 0753
VW 1156

petous22, 2018-02-04:
Confirm full working on Seat Leon 2017 (facelift 8") MIB2Std fw: 0462

IllClown, 2018-02-05:
Not working on Seat Leon 2016 MIB2Std 6,5" Hardware H28 FW 0351

marcobe, 2018-02-07:
Also not working on VW GTI Autobahn 2017 FW 0755 (USA)

petous22, 2018-02-08:
Confirm that version 0.36 + AAvexvag 2.3 (old one) full working on Seat Leon 2017 (facelift 8") MIB2Std fw: 0462, but version 0.37+AAvexvag 2.2 do not display data from car. All permisions for both application enabled. Full link data Exchange enabled too.

leymike, 2018-02-20:
Not working in HW 420 SW 617, Performance monitor display with no data. carlog folder empty.  Tested from .36 to .40b and aavexvag 2.0 to 2.2

lars18th, 2018-06-05:
Hi,

Tested with current version (0.45b + AAvexvag 2.2) inside a VW Golf 2016 HW:H24, SW:0430... fully white list in AA.

Results:
- The App works inside AA.
- The App doesn't receives any data from the car.
- The directory "/storage/emulated/0/CarLogs/" is empty.

So, please add **0430** to the list of **unsupported** HU.

Now, some questions:

1. Can you add some "testing" mode for checking the communication between the VEX-VAG capture service and this App? The idea is impersonate some fake values from the car and see that this App works correctly.
2. Please, add too one option to save the log in the External-SD.
3. The problem with unsupported HUs, is the user/pass in the AAvexvag App? What I need to do if I like to search the values of my HU? Any idea?

Regards.

lars18th, 2018-06-07:
Hi @mstuettgen ,

> Until then, is there any way to Offline-Test if the app is working correctly?

Sure!
Install the HeadUnit emulator:
https://play.google.com/store/apps/details?id=gb.xxy.trial
You can install it in one Android Tablet and then using one OTG cable connect to your mobile. It impersonates the car! It works without root.
And if your phone is rooted, then you can install it and use on it at the same time.

In any case, other HU emulators exists for Pi... However, for basic testing this is the most simple solution.

The only missing function until now is the "demo-fake-vex" App. We hope soon to have it.

Regards.

lars18th, 2018-06-07:
Hi @jilleb ,

>> The Whitelist App needs to be the "mqb-pm" or the "aauto-vex-vag"? I suspect is the second one, right?

> The whitelisted app is the service that communicates with the aauto-vex-vag.

You say then that the "aauto-vex-vag" needs to be in the white-list of the Head Unit (MIB2) of the car?
I suspect this can be done only using the correct exlap user/pass... right?

Please, explain more as I don't know where to search to enable access to my MIB2 HU.

Regards.

mstuettgen, 2018-06-07:
@lars18th :
< Sure!
< Install the HeadUnit emulator:
< https://play.google.com/store/apps/details?id=gb.xxy.trial
< You can install it in one Android Tablet and then using one OTG cable connect to your mobile. It impersonates the car! It works without root.
< And if your phone is rooted, then you can install it and use on it at the same time.

Neither is my phone is rooted (yet) not to I have a second tablet. Any chance there is a windows emulator or anything for that stuff?

@jilleb : 

< The app should not crash upon starting. Do not install multiple aauto-vex-vag releases at the same time. Only install 1. If one doesn't work, try an other version.

I managed to install aauto-vex-vag with ADB Fastboot, now there is no more error show when I start the configuration tool.

I also found some guides how to enable custom apps in Android Auto >= 3.0 (using ADB Fastboot), have you tried that yet? Otherwise I might give it a shot. 

Further more, could you use some help regarding the german translations for the app? 

best regards

mstuettgen, 2018-06-07:
Hey guys, I had to re-install my phone after I messed up rooting it, and now the vex-vag-2.2 doesnt seem to work anymore? How can I uninstall it? It's not listed under "apps" and I dont know the name of the package to uninstall it with adb shell. Any hints on that?

mstuettgen, 2018-06-08:
And I need help downgrading google services ... any helpful literature on that available?

lars18th, 2018-06-08:
> Neither is my phone rooted (yet) not do I have a second tablet. Any chance there is a windows emulator or anything for that stuff?

Sure! You don't need at all to root your device. And if you don't have a car with AA support, you can use the good HeadUnit emulator. And in case you don't have a second Android device, then you can use any Android PC Emulator, like the good MEmu: http://www.memuplay.com/

mstuettgen, 2018-06-08:
> Sure! You don't need at all to root your device. And if you don't have a car with AA support, you can use the good HeadUnit emulator. And in case you don't have a second Android device, then you can use any Android PC Emulator, like the good MEmu: http://www.memuplay.com/

@lars18th: thanks for the advice and the helpful tips. I tried a lot yesterday and today, but I ran into several problems.

Szeneario 1: Headunit Reloaded and Performace/Monitor on my (finally rooted) Android Phone

I managed to install the aauto-vex-vag2.2 and PM450B as well as Android Auto 2.9 and Headunit Reloaded at the same time. When I start up the Headunit-Reloaded and the performance monitor, Android Auto automatically starts (I guess that is a good sign!?). Further more my phone informs me that the headunit server is running, but the "screen" of the simulated headunit stays black....

Problably thats because I didn't downgrade Google Play Services, but I wasn't able to figure out how to do that. And since I didnt wanna screw up my phone again, I decided to set up Szenario 2.

Szenario 2: Headunit Reloaded on a virtual device (genymotion) without any google default apps
I downloaded genymotion and installed a bare android on a virtual device. Then I downloaded the APK for Google Play Services 11.9 (was the hights below 12.1 that I could find). After that, I uploaded the APK for Headunit Reloaded and GPS 11.9 onto the virtual device and installed them with "adb shell pm install....", both reporting "success". But when I tried to start up the headunit reloaded, the app just crashes....

Both Szenarios really suck and I am totally frustrated, but I really want to get that stuff going. Any advice for android noobs? Maybe some literature about how to set the headunit up with http://www.memuplay.com/ ?

best regards

lars18th, 2018-06-08:
Hi @mstuettgen,

- Scenario 1: In this case, the problem is because with the "Self Mode" you don't have started the AA Developer server. Read carefully the guide of the HU Reloaded. In fact, I never use the "Self Mode".

- Scenario 2: You can install *any* compatible version of GPS in the device where you run the HU Emu. If the App crash, then the problem is the Android Emu.

For the MEmu, check any guide about how to install on it GPS. Then use the HU Reloaded.
Another option is install a pure Android x86 in a Virtual Machine in your PC, and run the HU on it.
Or use the official DHU Emulator from Google: https://developer.android.com/training/auto/testing/

You have a lot of options! At time the problem isn't to have a Head Unit compatible with Android Auto.

jilleb, 2018-06-08:
@all, please stick to the topic, keep Google Play Services issues and questions in the right issue, #27 ,otherwise it's hard for me to keep track of all issues and questions.

> You say then that the "aauto-vex-vag" needs to be in the white-list of the Head Unit (MIB2) of the car?
I suspect this can be done only using the correct exlap user/pass... right?
Please, explain more as I don't know where to search to enable access to my MIB2 HU.

On the unit, there's a executable, "gal", Google Automotive Link. This executable has a config file, gal.json. Older versions/firmware distributions have a gal version without whitelist entry in this file, and the binary doesn't have any whitelist references either, so there will be NO communication of exlap data to Android Auto. No passwords help, there's no communication allowed from AA to the exlap communication channel of the unit on these versions. That's the main problem. I tried replacing the gal and gal.json files with more recent versions, but since the binairies are compiled using a different codebase, this won't work. Just the config file didn't work either, as the binary doesn't accept any whitelist data, since it's not looking for it.

> I also found some guides how to enable custom apps in Android Auto >= 3.0 (using ADB Fastboot), have you tried that yet? Otherwise I might give it a shot.

Feel free to try them, I can't support these, but maybe someone else can.

>Further more, could you use some help regarding the german translations for the app?

Yes, yes yes :-) my German isn't the best, so if you want to help out, please make a pull request on the german language file with your suggested changes.

jilleb, 2018-12-03:
Wow, I didn't expect it to work with vex vag v2.4 already  😄 
Good to hear that, thanks!

MisterNo1, 2018-12-06:
After a several days of using, unfortunately I noticed that the issues are here.
After a cca 2 minutes of use, the app has a small lag, than freeze, and at the end, totaly exit to main AA menu  :(