jilleb/mqb-pm #2: Adding data to one of the 4 textviews makes the app crash

jilleb, 2018-02-03:
Some users report: When _data_ (currently not known what specific data) is added to one of the 4 textviews, the app crashes.

More info is needed, but I'm looking into this.

jilleb, 2018-02-07:
Probably fixed in the latest release. It turned out the USA units don't share their temperature-units in the same way as European units do.

Calfredop, 2018-02-13:
Still occuring in 0.39b

Calfredop, 2018-02-13:
Sorry forgot to mention. The app crashes when there's data to show, for instance when temperature hit threshold or when more dynamic data is choosen like RPM.

jilleb, 2018-02-13:
All right. Can you make a logcat of that?

Calfredop, 2018-02-13:
`02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: Process: com.mqbcoding.stats, PID: 3733
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: java.lang.ClassCastException: java.lang.Float cannot be cast to java.lang.String
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment.updateElement(DashboardFragment.java:1093)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment.doUpdate(DashboardFragment.java:482)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment.access$500(DashboardFragment.java:40)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment$4.run(DashboardFragment.java:467)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:790)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6494)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
02-13 18:51:04.650 10285  3733  3733 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)`

This is just a snippet, the log is full of ClassCastExcetions from various PID. If I may, you could log the incoming data for better debug.

jilleb, 2018-02-13:
Thanks, this is valuable!
I'll add some more logging for debugging. 

It's strange...
mLastMeasurements.get(queryElement) should be a string... Buy maybe I'm wrong. I'll experiment some more and report back.

jilleb, 2018-02-14:
@Calfredop, should be fixed in v0.4

Calfredop, 2018-02-14:
Hi,
thanks for your quick response but now we got a new exception.
`02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: FATAL EXCEPTION: main
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: Process: com.mqbcoding.stats, PID: 4904
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: java.util.IllegalFormatConversionException: f != java.lang.String
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at java.util.Formatter$FormatSpecifier.failConversion(Formatter.java:4402)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at java.util.Formatter$FormatSpecifier.printFloat(Formatter.java:2897)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at java.util.Formatter$FormatSpecifier.print(Formatter.java:2844)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at java.util.Formatter.format(Formatter.java:2523)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at java.util.Formatter.format(Formatter.java:2458)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at java.lang.String.format(String.java:2814)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment.updateElement(DashboardFragment.java:1144)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment.doUpdate(DashboardFragment.java:482)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment.onCreateView(DashboardFragment.java:371)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.Fragment.performCreateView(Fragment.java:2354)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1419)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1740)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.BackStackRecord.executeOps(BackStackRecord.java:794)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.FragmentManagerImpl.executeOps(FragmentManager.java:2580)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2367)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2322)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:2229)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3221)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:3171)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.support.v4.app.FragmentController.dispatchActivityCreated(FragmentController.java:192)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.google.android.gms.car.e.onStart(Unknown Source:22)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.google.android.apps.auto.sdk.CarActivity.onStart(Unknown Source:0)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.mqbcoding.stats.MainCarActivity.onStart(MainCarActivity.java:254)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:458)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:210)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:595)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.google.android.gms.car.internal.i.run(CarActivityHostImpl.java:2)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:790)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6494)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
02-14 18:42:12.400 10286  4904  4904 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
02-14 18:42:12.405  1000  1140  1706 I DisplayManagerService: Display device removed: DisplayDeviceInfo{"com.mqbcoding.stats/com.mqbcoding.stats.MainCarActivity": uniqueId="virtual:com.mqbcoding.stats,10286,com.mqbcoding.stats/com.mqbcoding.stats.MainCarActivity,0", 800 x 400, modeId 289, defaultModeId 289, supportedModes [{id=289, width=800, height=400, fps=60.0}], colorMode 0, supportedColorModes [0], HdrCapabilities null, density 160, 160.0 x 160.0 dpi, appVsyncOff 0, presDeadline 16666666, touch NONE, rotation 0, type VIRTUAL, state ON, owner com.mqbcoding.stats (uid 10286), FLAG_PRIVATE, FLAG_NEVER_BLANK, FLAG_PRESENTATION, FLAG_OWN_CONTENT_ONLY}`

Perhaps is due to the fact that I'm using VAG service 2.1?

jilleb, 2018-02-14:
I'll have a look into this tomorrow. I still haven't had time to install my unit back into the car, so can't test it in the real world..

jilleb, 2018-02-14:
@Calfredop , I just took a look at the example application. The dev did it like this:

`        Float oilTemp = (Float) mLastMeasurements.get("oilTemperature");
        if (oilTemp == null) {
            mOilTemp.setText("----");
        } else {
            mOilTemp.setText(String.format(Locale.US,
                    getContext().getText(R.string.temp_format).toString(), oilTemp));
        }`

While my code: 
`            case "vehicleSpeed":
                Float mVehicleSpeed = (Float) mLastMeasurements.get("vehicleSpeed");
                String speedUnit = (String) mLastMeasurements.get("vehicleSpeed.unit");
                if (mVehicleSpeed != null && speedUnit != null) {
                    value.setText(String.format(Locale.US, getContext().getText(R.string.decimals).toString(), mVehicleSpeed));
                    label.setText(speedUnit);
                }`

In your logcat, things go wrong at line 1144, which is:
`                Float mVehicleSpeed = (Float) mLastMeasurements.get("vehicleSpeed");
`

I don't understand why it's going wrong.

Calfredop, 2018-02-14:
TBH I don't know, because I don't have a full sight on your code. But if you use a `Map<String, Object>` perhaps instead of casting to class you could try to use `.toString()` or for debug purposes `System.out.println(mVehicleSpeed.getClass().getName())` befor casting so you can be sure which class is the object in the Map.

Calfredop, 2018-02-14:
After taking a look at the code I think another good idea for debug is to log all data put to the mLastMeasurements at line 455
   `private final CarStatsClient.Listener mCarStatsListener = new CarStatsClient.Listener() {
        @Override
        public void onNewMeasurements(String provider, Date timestamp, Map<String, Object> values) {
            mLastMeasurements.putAll(values);
//Log the k, v to file
            postUpdate();
        }
    };`

Calfredop, 2018-03-06:
Now most of the data seems to work (ex. RPM, torque, power...) in the 4 views but temperatures still not working. I've tested v0.40b + VAG Service 2.2. Also when temperature data is set in one of the three clocks, it only shows zeros.
`03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: FATAL EXCEPTION: main
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: Process: com.mqbcoding.stats, PID: 28611
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: java.util.IllegalFormatConversionException: f != java.lang.String
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at java.util.Formatter$FormatSpecifier.failConversion(Formatter.java:4402)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at java.util.Formatter$FormatSpecifier.printFloat(Formatter.java:2897)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at java.util.Formatter$FormatSpecifier.print(Formatter.java:2844)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at java.util.Formatter.format(Formatter.java:2523)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at java.util.Formatter.format(Formatter.java:2458)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at java.lang.String.format(String.java:2814)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment.updateElement(DashboardFragment.java:1189)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment.doUpdate(DashboardFragment.java:482)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment.access$500(DashboardFragment.java:40)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at com.mqbcoding.stats.DashboardFragment$4.run(DashboardFragment.java:467)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:790)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6494)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
03-06 18:44:43.189 10295 28611 28611 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
03-06 18:44:43.201  1000  1115  2683 I DisplayManagerService: Display device removed: DisplayDeviceInfo{"com.mqbcoding.stats/com.mqbcoding.stats.MainCarActivity": uniqueId="virtual:com.mqbcoding.stats,10295,com.mqbcoding.stats/com.mqbcoding.stats.MainCarActivity,0", 800 x 400, modeId 21, defaultModeId 21, supportedModes [{id=21, width=800, height=400, fps=60.0}], colorMode 0, supportedColorModes [0], HdrCapabilities null, density 160, 160.0 x 160.0 dpi, appVsyncOff 0, presDeadline 16666666, touch NONE, rotation 0, type VIRTUAL, state ON, owner com.mqbcoding.stats (uid 10295), FLAG_PRIVATE, FLAG_NEVER_BLANK, FLAG_PRESENTATION, FLAG_OWN_CONTENT_ONLY}`

jilleb, 2018-03-06:
I tested my new code a few mins ago. The issue is with the temperature unit and the way I use it. Took it out, now everything works. Expect a new version soon!

Calfredop, 2018-03-06:
Little update, the battery voltage also makes the app crashes. Thanks for your follow-up.

jilleb, 2018-03-06:
Yes, basically everything that had a unit behind the value:
12V
89%
12G
67°

The latest code has most of these fixed :-)
New release will follow this week, after testing.

jilleb, 2018-03-20:
Closing, this issue is fixed as of version 0.42B:
https://github.com/jilleb/mqb-pm/releases/tag/v0.42b-beta

closed 2018-03-20 by jilleb