jilleb/mqb-pm #5: Building from source not possible without google-services.json

nerone-github, 2018-02-07:
Hi, i just tried to build from source, however it is not possible to build correctly since the google-services.json file is missing.

Please provide instructions how it is possible to build from source :) 

[image]

jilleb, 2018-02-11:
The google-services.json was left out, because it contains my API key, and I don't really like the idea of others using my api key ;-) But.... you can easily make your own, check the answers to this:
[stackoverflow topic](https://stackoverflow.com/questions/34367870/where-do-i-get-a-google-services-json)

About the missing drawable: I'll fix that with a new commit today! my bad ;-)

jilleb, 2018-02-11:
Fixed the missing drawable in the latest commit.

closed 2018-02-11 by jilleb