jilleb/mqb-pm #9: Will not work with Android Auto V3

jilleb, 2018-02-17:


PhilippMundhenk, 2018-02-27:
Could you give a few more details? Is it possible that the app doesn't show up in the OEM menu, despite "Unknown Sources" enabled?
I seem to have this issue for a few days now with my own app. It was working great formerly both in car and simulation, but suddenly stopped working in the car. Still works fine in the simulator. Can you confirm that this is related to v3?

jilleb, 2018-02-27:
Yes, I can confirm this. On xda-developers in Android Auto topics, several devs have the same issue. It works in the simulator, but in a real car it will not work.

manudroid19, 2018-10-09:
I have almost ready a root app that will enable any app in Android Auto through an easy UI. It will be published in XDA in a few days. 

Also, to keep list of firmwares updated,  while testing the app, I tested this app along others, and my car isn't sending any data :(

Last-gen Audi, firmware 0213

jilleb, 2018-11-16:
Linking @manudroid19 's app here:

https://forum.xda-developers.com/android-auto/android-auto-general/app-android-auto-3rd-party-app-enabler-t3853009

It works like a charm! 👍 👍

closed 2018-11-16 by jilleb