jilleb/mqb-pm #20: App slows down over time, and then restarts

jilleb, 2018-03-30:
The app seems to slow down over time, (15 minutes or so), and then restarts.

kaisertgc, 2018-03-30:
I have the same issue also. It is getting data, next time it doesnt.

kaisertgc, 2018-03-30:
[car-20180330183328.log](https://github.com/jilleb/mqb-pm/files/1864816/car-20180330183328.log)

jilleb, 2018-03-30:
I think there's some memory leak, or it's the graph causing troubles..

javigody, 2018-03-30:
@jilleb I have the same problem, also before de the graph implementation.

javigody, 2018-04-02:
Maybe implementing vex-base version 4.3 the problem gets solved

jilleb, 2018-04-03:
@javigody , I tried doing that, but currently it brings me several errors:
[image]

Probably because of 03516b0c8b1477870460cd8b67091075cc9b2409

javigody, 2018-04-03:
@jilleb I have solved half of the problems. I keep investigating on the other half.

jilleb, 2018-04-04:
Great, thank you! :-)

javigody, 2018-04-04:
Needs revision, no data output. The app still crashes over time.

jilleb, 2018-04-13:
I added a start/pause for the graph, should also help a little. Otherwise it would be gathering data all the time.

jilleb, 2018-04-18:
I added a delay with commit eef3a04b4b147d3b475554fcc1d17900401f4998. It used to post an update as soon as new measurements are available.. which can be VERY often, and I assume this can cause performance issues after a few minutes. I'll test it today, let's hope my assumption is right.

javigody, 2018-04-18:
Great! I'm going to try it out right now.

jilleb, 2018-04-18:
Word of warning: i did a lot of changes to the measurement fragment, I'm not done with it, which will probably cause a crash ;-)

jilleb, 2018-04-18:
Changes seem to be successful! No more freezes and crashes.

javigody, 2018-04-23:
Well, after testing a bit more, it takes more time to crash, maybe 10 minutes, but it still crashes and freezes.

jilleb, 2018-07-03:
@javigody , I've updated the code with all changes @martoreto did in AAstats. It looks like the app isn't receiving any data now. Gonna test some more tonight.

divacky, 2018-12-10:
Hi Chillout,

I've made some changes to the app.
1) I moved the bindService method from onAttach to onResume and unbindService from onDettach to on
Pause, both in DashboardFragment and GraphFragment. This way, the method is only run once and the application will not freeze. I’m testing for the third day without freezing.
2) I fixed the "element_currenttorque" element as "currenttorgue" which works.
3) Added two more elements "shortTermConsumption" and "NAV_CurrentPosition.street".

Could you look at it and eventually to integrate into the project? The modified files attached.
[commit.zip](https://github.com/jilleb/mqb-pm/files/2665016/commit.zip)
Divacky

jilleb, 2018-12-11:
Good work @divacky 👍 I will put it in today and test

jilleb, 2018-12-11:
90cae947baebace08defc75c8d0818d3c82a8302 has your changes. I didn't include the graphfragment changes yet, since I removed it. If this works, I will put it back 😄

divacky, 2018-12-11:
Hi,

GraphFragment I‘m not testing now, I‘m focusing on the DashboardFragment. Today I have driven over 150 km and in various modes and I have no problem. Stability is depends on letting Listner run only once. Therefore, I have moved the bindService method to onResume and onPause because this method is called only when the fragment is activated. The onAttach method calls right at the start of the application and runs in both the DashBoardFragment and GraphFragment. This step contributes to the stability of the application. Unchecked changes in GraphFragment will not have the desired effect because the service will run twice. You need to transfer code or comment bindservice in graphFragment.

Divacky

jilleb, 2018-12-11:
Yes, I totally removed graphfragment a few commits ago. I might put it back now with your fix applied 😎

divacky, 2018-12-12:
OK, I have installed. There is one error in the application, the unbind.Service method has remained in onDettach and causes an error when quitting. I removed this error for my today´s testing.

jilleb, 2018-12-12:
Ok, I will fix it :-)

divacky, 2018-12-12:
Today's testing has not made any mistakes and slowdowns. While running the app, I used phone calls, navigation, day / night mode and everything works correctly. The service runs properly in the background and after returning to the application it goes on..
I tested the menu item's change to StopwatchFragment and was stopped the service. After returning to the item in the DashBoardFragment menu, the service was started (this was reflected by resetting the min/max values).
Finally, the app works correctly while driving, and the phone is not busy.

MisterNo1, 2018-12-19:
Hi. I was download version 0.5B last week and I had a bag (slowdown app speed and than turn off). Is this last fix implemented in the v. 0.5B or not? Thanks

jilleb, 2019-01-17:
All is fixed as of 0.6B. 
Just make sure you have no app killers/battery savers or whitelist this app and the vex-vag-service.
Closing this issue :-)

closed 2019-01-17 by jilleb