jilleb/mqb-pm #25: Czech languauge update

divacky, 2018-04-05:
<?xml version="1.0" encoding="utf-8"?>
<resources>
    <string name="app_name">Monitor výkonu pro VAG</string>
    <string name="temp_formatF">%.0f°F</string>
    <string name="temp_format">%.0f°</string>
    <string name="car_service_name">Monitor výkonu</string>
    <string name="temp_formatC">%.0f°C</string>
    <string name="volt_format">%.1fV</string>
    <string name="gforce_format">%.1fG</string>
    <string name="percent_format">%.1f</string>
    <string name="degrees_format">%.1f°</string>
    <string name="decimals">%.1f</string>
    <string name="no_decimals">%.0f</string>
    <string name="psi">psi</string>
    <string name="bar">bar</string>
    <string name="zeroVolt">0,0V</string>
    <string name="zeroCelcius">0,0°</string>
    <string name="kmh">kmh</string>
    <string name="nm">Nm</string>
    <string name="kw">KW</string>
    <string name="activity_settings_title">Nastavení Monitoru výkonu</string>
    <string name="pref_appearance">Vzhled</string>
    <string name="pref_highvis">Reflexní číselníky</string>
    <string name="pref_theme">Téma</string>
    <string name="pref_highlow">Nejvyšší / nejnižší hodnoty</string>
    <string name="pref_accbrakeindicator">"Indikátor zrychlení/brždění "</string>
    <string name="pref_choose_account_title">Vyberte účet</string>
    <string name="pref_upload_to_bigquery_title">Odeslat logy do BigQuery</string>
    <string name="pref_bigquery_project_id_title">ID projektu</string>
    <string name="pref_bigquery_dataset_title">Jméno datasetu</string>
    <string name="pref_bigquery_table_title">Tabulka</string>
    <string name="pref_kick_uploads_title">Odeslat data</string>
    <string name="kick_uploads_nothing_to_do">Všechna data jsou odeslána.</string>
    <string name="kick_uploads_done">Odeslání dat naplánováno.</string>
    <string name="pref_bigquery_reupload_last_title">Znovu odeslat poslední log.</string>
    <string name="bigquery_reupload_last_failed">Selhalo odeslání posledního logu.</string>
    <string name="bigquery_reupload_last_ok">Opětovné odesílání dat.</string>
    <string name="oil_notification_title">Teplota oleje</string>
    <string name="oil_notification_text">Motorový olej dosáhl provozní teploty.</string>
    <string name="pref_oil_temp_threshold_title">Pracovní teplota oleje</string>
    <string name="pref_oil_temp_monitoring_title">Monitorování teploty oleje</string>
    <string name="pref_oil_temp_monitoring_summary">Oznámit dosažení provozní teploty oleje.</string>
    <string name="pref_notifications_category_title">Oznámení</string>
    <string name="pref_wheel_state_monitoring_title">Pípnout při nulovém úhlu volantu</string>
    <string name="pref_wheel_state_monitoring_summary">Pípne pokud při zařazené zpátečce je volant v nulovém úhlu.</string>
    <string name="pref_logging_category_title">Logování</string>
    <string name="pref_stats_logging_title">Log s daty z auta</string>
    <string name="pref_stats_logging_summary">Složka s logy: %1$s</string>
    <string name="pref_stats_logging_not_available">Logování není dostupné: %1$s</string>
    <string name="pref_providers_category_title">Poskytovatel telemetrických dat</string>
    <string name="pref_list_providers_title">Seznam poskytovatelů</string>
    <string name="providers">Poskytovatel telemetrických dat</string>
    <string name="pref_leftclock">Levý budík</string>
    <string name="pref_centerclock">Prostřední budík</string>
    <string name="pref_rightclock">Pravý budík</string>
    <string name="pref_view1">Náhled 1</string>
    <string name="pref_view2">Náhled 2</string>
    <string name="pref_view3">Náhled 3</string>
    <string name="pref_view4">Náhled 4</string>
    <string name="pref_pressureunit">Jednotka tlaku</string>
    <string name="pref_dataelementsettings">Nastavení datových prvků</string>
    <string name="pref_highvis_off">Zobrazit ručičku na budících</string>
    <string name="pref_highvis_on">Zobrazí reflexní číselníky</string>
    <string name="pref_highlow_off">Nezobrazí minimální a maximální hodnoty</string>
    <string name="pref_highlow_on">Zobrazí minimální a maximální hodnoty</string>
    <string name="pref_accbrakeindicator_off">Indikátor je zobrazen</string>
    <string name="pref_accbrakeindicator_on">Indikátor není zobrazen</string>
    <string name="pref_highlowmarks">Značka min/max hodnoty na budících</string>
    <string name="pref_highlowmarks_on">Minimální a maximální hodnoty jsou zobrazeny.</string>
    <string name="pref_highlowmarks_off">Minimální a maximální hodnoty nejsou zobrazeny.</string>
    <string name="pref_graphitem">Graf</string>
    <string name="graphDelay">Rozlišení grafu</string>
    <string name="volt">V</string>
    <string name="g">G</string>
    <string name="mph">mph</string>
    <string name="rpm">ot.</string>
    <string name="activity_stopwatch_title">Stopky</string>
    <string name="activity_measurements_title">Měření</string>
    <string name="activity_graph_title">Grafy</string>

    <string name="none">Nepoužito</string>
    <string name="absChargingAirPressure">Tlak turba (absolutní)</string>
    <string name="relChargingAirPressure">Tlak turba (relativní)</string>
    <string name="batteryVoltage">Napětí baterie</string>
    <string name="coolantTemperature">Teplota chladící kapaliny</string>
    <string name="oilTemperature">Teplota oleje</string>
    <string name="gearboxOilTemperature">Teplota převodovky</string>
    <string name="outsideTemperature">Venkovní teplota</string>
    <string name="vehicleSpeed">Rychlost</string>
    <string name="currentGear">Zařazený rychlostní stupeň</string>
    <string name="recommendedGear">Doporučený rychlostní stupeň</string>
    <string name="engineSpeed">Otáčky motoru</string>
    <string name="lateralAcceleration">Příčné přetížení</string>
    <string name="longitudinalAcceleration">Podélné přetížení</string>
    <string name="yawRate">Náklon</string>
    <string name="wheelAngle">Úhel volantu</string>
    <string name="acceleratorPosition">Poloha plynového pedálu</string>
    <string name="brakePressure">Tlak na brzdový pedál</string>
    <string name="powermeter">Hybridní výkon</string>
    <string name="currentTorque">Kroutící moment</string>
    <string name="currentOutputPower">Výstupní výkon</string>
    <string name="EcoHMI_Score.AvgShort">Eko skóre</string>
    <string name="EcoHMI_Score.AvgTrip">Eko skóre průměr</string>
    <string name="currentConsumptionPrimary">Aktuální spotřeba, primární palivo</string>
    <string name="currentConsumptionSecondary">Aktuální spotřeba, sekundární palivo</string>
    <string name="cycleConsumptionPrimary">Spotřeba od načerpání, primární  palivo</string>
    <string name="cycleConsumptionSecondary">Spotřeba od načerpání, sekundární  palivo</string>
    <string name="vehicleSpeed_alternative">Ryclost (jiný zdroj)</string>
    <string name="test">Test (zobrazí náhodná data,která nepochází z auta)</string>

    <string name="author">Aplikaci vytvořil:</string>
    <string name="basedon">Na základě SDK a demo aplikací od:</string>
    <string name="thanks">Speciálně děkuji:</string>
    <string name="drivesafely">Bezpečnou jízdu.</string>
    <string name="testing">testování</string>
    <string name="activity_credits_title">O projektu</string>
    <string name="activity_main_title">Monitor výkonu</string>
    <string name="activity_carstatus_title">Info o autě</string>
    <string name="resettoapply">Změna nastavení se projeví opětovným připojením telefonu k systému Android Auto.</string>
</resources>

jilleb, 2018-04-05:
Thank you so much! 👍 

I implemented the new strings with 0c2e6edac2d3ed427f00dd00ea21631cc6132a62

closed 2018-04-05 by jilleb

divacky, 2018-04-17:
I'm sending corrections to Czech translation.
Please repair the output power unit. Displays Nm and should be Kw.
[strings_czech.zip](https://github.com/jilleb/mqb-pm/files/1921373/strings_czech.zip)

jilleb, 2018-04-17:
Thanks, implemented the language with 41f2afc42b41b5d36eb3ec5d35a13bdd19a2e3a6, Output power as Kw with 1f542a1e46667594f470d232f8173bf9d048e9a1 :-)

divacky, 2018-12-03:
I'm sending corrections to Czech translation.
[strings_CS.zip](https://github.com/jilleb/mqb-pm/files/2640867/strings_CS.zip)

jilleb, 2018-12-03:
Thank you 👍 

implemented your latest strings with #6daaa004ee8e5ae54c149a66cae30f8fc8f5d50e

divacky, 2019-01-17:
I'm sending new corrections to Czech translation.
[strings_cs_20190117.zip](https://github.com/jilleb/mqb-pm/files/2770825/strings_cs_20190117.zip)

jilleb, 2019-01-20:
Thanks! Added the new strings.