jilleb/mqb-pm #27: Custom applications not available on AA after Google Play Services 12.5.21

jilleb, 2018-04-06:
Since the latest google play services, custom apps are fully disabled... This means custom apps will no longer show up in Android Auto, which prevents this app from being used 👎 😠 😡 :rage1:

jilleb, 2018-04-09:
Currently in the process of publishing the app to Google Play.

jilleb, 2018-04-10:
Aaaaaaaaaaand it's rejected 👎 
Maybe if I publish it as "internal test", it's more succesful, but then I'm limited to 100 users.

jilleb, 2018-04-11:
Internal test wasn't succesful either. I wrote an email to the reviewers, which was answered with:

> The Android Auto FullScreen SDK that you are using for your application is only available for use by Automotive OEM partners that have signed agreements with Google. Seat is one of these partners.  Accordingly, I'm sorry but we can't review your app for inclusion in the play store as you are not authorized to use the Android Auto FullScreen SDK.

jilleb, 2018-04-18:
Since this is an issue that can't be fixed by me, unless I decide to start a car brand, the easiest workaround is:

- roll back to an older Google Play Services version
- stay on Android Auto before V3.

I won't go into detail how to achieve this, since it's different for each phone/rom/situation. Use google :-)

setterlee, 2018-04-22:
if the app don't work in Android auto V3 with recent google play service, that's mean the end of this project? because deal with the auto update of google play service is really awkward... 

Is curious that the google's Team mention the Automotive Partners... Spotify have full screen, Waze too... xD

jilleb, 2018-04-22:
I'm continuing the project, since there are workarounds to get it working.

Also, Waze is owned by Google. Spotify uses the media SDK, which is allowed by Google. We can only use the Media and Messaging SDK's.

divacky, 2018-07-06:
I added the code that used thekirankumar in app carstream to unblock the application and which works on the current version of android auto 3.3.582063 and services google play 12.6.85. It is necessary to have root phone, but it is better than to still install older versions of Google Play services.  Could you please look at it and eventually to integrate into the project? Updated files attached.
[unlock.zip](https://github.com/jilleb/mqb-pm/files/2170898/unlock.zip)

jilleb, 2018-11-16:
There's now an unlocker-app:

https://forum.xda-developers.com/android-auto/android-auto-general/app-android-auto-3rd-party-app-enabler-t3853009

Use this, it works lovely!

closed 2018-11-16 by jilleb