jilleb/mqb-pm #28: VIN isn't always showing in Cardatafragment

jilleb, 2018-05-11:
The car shares the VIN only once in the entire communication session. If the connection is broken for some reason, the VIN won't be found in the latest measurements, so the value will remain null. 

Maybe it's an idea to save the VIN and other useful data to sharedpreferences whenever possible.

divacky, 2018-05-15:
Hi,
 
I have a car octavia. The performance monitor is now running, it starts to slow down until it crashes. My mobile phone is busy and hot.
I have the source code in ntb, can I test somehow what causes this problem? Carlogs after crashed up are unreadable.
 
Thanks
 
divacky

jilleb, 2018-05-16:
That is a different issue: #20 
I think it's caused because the car sends a LOT of data, so it takes more effort each time it has to get the latest measurements, since the pool of data is growing each time.

mstuettgen, 2018-06-07:
Hey there,

I am going to try it out in my Golf VII GTI (2016) with HW 820 and Software 0624 and let you know if it worked. Unfortunately I have to wait until next week for app-connect beeing activated by the VW service....  

Until then, is there any way to Offline-Test if the app is working correctly? 

I installed both latest apks from here:
https://github.com/martoreto/aauto-vex-vag/releases
and
https://github.com/jilleb/mqb-pm/releases

but all i can start on my phone now is the configuration tool for the performace monitor, which throws an exception that says "vag extensions have been stopped repeatedly (or sth similiar...). I guess thats because I am not connected to the car, right?

Further more, when I start the Android-Auto-App (V2.9....something) in developer mode with untrusted sources allowed, I dont see the performance monitor app. Is this also because I am not connected to the car?

Thx and best regards