jilleb/mqb-pm #31: New header

divacky, 2018-12-19:
in the app I would like to have a header as it is in original navigation (I'm sending a picture). that is, add the option to display the current location (street, city) and outside temperature in the settings. I wanted to make graphic changes but I do not have access to Martoret's SDK Auto (private use case). Can it be made available?
divacky
[image]

divacky, 2018-12-31:
Hi,
Data from the navigation are displayed incorrectly, it can not display characters á, í, etc.
[image]

jilleb, 2019-01-08:
Ahh thanks, good ones. I'll get on this asap. I had some computer issues, but I'm (almost) back!

jilleb, 2019-01-09:
> Hi,
> Data from the navigation are displayed incorrectly, it can not display characters á, í, etc.

Is this happening with all fonts? I think it's caused by the font not supporting those characters, I will have to look for an extended version of that font

> in the app I would like to have a header as it is in original navigation (I'm sending a picture). that is, add the option to display the current location (street, city) and outside temperature in the settings.
> divacky
This one is fixed with d31929c

divacky, 2019-01-10:
Hi,

yes, it happens with all the fonts. If I display a string instead of the navigation data, it will appear correctly. The characters are displayed incorrectly: á, í, é, no, but they are displayed correctly, for example: Ř, ž, š.
[image]

A week ago I made another view of navigation data.
[image]

jilleb, 2019-01-10:
> yes, it happens with all the fonts. If I display a string instead of the navigation data, it will appear correctly. The characters are displayed incorrectly: á, í, é, no, but they are displayed correctly, for example: Ř, ž, š.

Okay, I will have a look at that. :-)

> A week ago I made another view of navigation data.

That looks very good and OEM-like! Much better than what I'm currently using, centered in the top (see red arrow).

[image]

Did you put a textview in the top section of the layout?

divacky, 2019-01-13:
Hi,
yes I used a textview, just reduced the font size at the location field. If the outdoor temperature drops below 4 degrees, this temperature is displayed in red.  I do not know if it's practical. Your changes look good. Today I will test it in the car.

jilleb, 2019-01-13:
Thanks! I've noticed the touch-the-clock-to-switch-to-graph feature isn't as fluent as I'd hoped, but I'll fix that soon. Currently struggling with a broken laptop, which slows down development a little.

divacky, 2019-01-29:
I found that the file schema.json contains bad characters. So it looks like all exlap communication is wrong.
[image]

divacky, 2019-01-29:
New Header 2. I added a distance and time to the destination,  and added a line
[image]

memeller, 2019-02-01:
> 
> I found that the file schema.json contains bad characters. So it looks like all exlap communication is wrong.
> [image]

I have checked the vex-vag part and it seems for me that the problem with special characters missing is located in this module. If you check the byte buffer that contains the bytes supplied by the car you can clearly see that there is some extra information. In my case ("Łódź" string) the byte buffer has the corresponding data: {-59, -127, 32, -77, 100, -59, 32} (signed ints). All of the special characters are encoded using two bytes (Ł is '-59,-127'; ó is '32,-77', ź is '-59,32') and normal ASCII characters are using only one byte (d is '100'). This proves that since the codes are different, the characters are present in the data provided by car. 
Data from your screenshot was also generated by vex-vag module so it is no suprise that it also shows some of the characters missing.

divacky, 2019-02-11:
It is so. I have found that utf-8 characters beginning with c3 and c4 are incorrect, but the characters beginning with c2 and c5 are correct. I could not convert characters in byte [] from exlap to some character set that would be correct. It will need to be replaced in source data from exlap.
[image]

divacky, 2019-02-18:
EXLAP is a bad way for diacritics. The new way
 is google which I do not need navigation in the car. I made a test apk who is interested in testing. @Chilout has not been here for 25 days ...
[Performance_70B_test.zip](https://github.com/jilleb/mqb-pm/files/2877074/Performance_70B_test.zip)

memeller, 2019-02-28:
I've checked your test version and noticed the following:
- both temperature and location are shown when there is active internet connection. If there is a connection then a white globe is shown in top left corner. Maybe add red globe icon like in car's original ui to mark that the connection is not available?
- the special characters are displayed properly
- for some reason the length of location text was limited to 4 chars. I am not sure if this is anyhow connected with the name of the city where I am that also has the length of 4 chars? The city was displayed properly but then all of the street names were trimmed to only 4 first chars.

As for the exlap encoding it would be suprising if a German company (that uses language specific chars on a daily basis) would skip checking the proper display of such data. Especially if the Exlap pdf that we all have seen specifies utf-8 as selected encoding for the whole system. I want to check if seat's driveapp vex service approach to connection differs in any way from the vex vag that we are using. There are no signs of location based data being shown in the aa app interface, but maybe the connecting service works differently.

divacky, 2019-02-28:
thanks for testing. I'm currently testing location by using googleapi. 
It looks the most accurate, unfortunately this way has problems too. E.g. if 
the roads cross, the street is shown from another position. Entrance into the 
tunnel changes the position above the tunnel, 
 
I fixed displaying of 4 characters, an error in the textview property.

divacky, 2019-03-02:
A new test version that uses googleApi. It is necessary to have enabled a GPS on your mobile
[Performance monitor 70D_test.zip](https://github.com/jilleb/mqb-pm/files/2922045/Performance.monitor.70D_test.zip)

divacky, 2019-06-19:
My version of what I use in my car. I have fixed some EXLAP characters (á, č, ř, ž, etc.) Unfortunately, some do not go because EXLAP sends only 2 spaces. (í, é, ě, etc.)
I adjusted the look of the screen. I try to display street and city names from Exlap (more accurate than Google), if the name contains a valid character, I will display the Google GPS data. I removed the min and max values I moved to the bottom of the clocks. I have expanded to 3 screens dataelements 1 - 3, which can be switched using the buttons on the screen and I can define a name for each screen. Now I'm trying to return the 4 graph screen, sharing the performance monitor data for the other screens. I'm programming this now. see picture.
[image]

jilleb, 2019-07-30:
I think the header will need major rework with the new AA 4.5 GUI. Closing this issue.

closed 2019-07-30 by jilleb