jilleb/mqb-pm #35: Location info from Google maps if exlap location info isnt available.

jilleb, 2019-01-18:
Currently, there is no location for people without exlap data. Make a fallback to Google maps location.

mobrembski, 2019-07-30:
I've created something simmilar:
https://github.com/jilleb/mqb-pm/pull/56
It uses basic android geocoding from location API to solve street name/street number/city

jilleb, 2019-07-30:
Ahhh nice :-)
The same feature was implemented by @Lucas-M-G the other day , I'll have a look to see I can merge your code in a safe way

mobrembski, 2019-07-30:
Ahh, right, i haven't noticed that you merged Lucas-M-G changes :(
I will take a look at it, but as i can see he added an display of navigation instructions, i think this might be a different feature.
Hold on, i will try to rebase my patches.

Lucas-M-G, 2019-07-31:
My changes only display the title of the Google Maps notification. If there is no active navigation, the current Street was displayed otherwise the next instruction. This solution doesn’t need any extra requests, but it is not the cleanest solution.

jilleb, 2019-11-14:
Closing issue, I believe it's done :-)

closed 2019-11-14 by jilleb