jilleb/mqb-pm #45: Add multiple-panel layout

xDefcon, 2019-06-03:
Hello, 
I want to suggest the option to have multiple panels, as the current interface can contain a maximum of 3 graphical "indicators"  (with relative graphs by clicking on them) and four text-based views.

The multi-pane would be perfect if with a swipe to the left you could change the current pane with another one and have, for instance, twice as the information you want without configuring it in the app and having to disconnect/reconnect the phone.

I don't exclude the fact that I would be able in the future to open a pull request with the specified changes, I have to see the code of the Android Auto app.

xDefcon, 2019-06-20:
Similar to #41

divacky, 2019-08-15:
I made multiple layouts, but I am not allowed access to the new pullrequest. You can allow this?

xDefcon, 2019-08-21:
@divacky in the meantime, could you upload your commits to the repository you forked? Thanks.

jilleb, 2019-08-22:
> Can I grant access or should I do otherwise?

I checked the setetings, and everyone should be able to post a pull request.. strange stuff..

divacky, 2019-09-04:
Sorry, I'm busy working. I am sending the changed files in the attachment. Can you compare files with the current version? Thank you
[mqb.zip](https://github.com/jilleb/mqb-pm/files/3574986/mqb.zip)

divacky, 2020-01-01:
New pull request with multidashboard

jilleb, 2020-01-07:
Niiiiice @divacky .I'll merge them tomorrow. Thanks!

jilleb, 2020-01-09:
Merged, and it's very very cool :-)

closed 2020-01-09 by jilleb

xDefcon, 2020-01-09:
> Merged, and it's very very cool :-)

Plan to add a new release with that changes?

jilleb, 2020-01-10:
of course :-)