jilleb/mqb-pm #46: Add 2D graph that combines lateral and longitudinal acceler

xDefcon, 2019-06-03:
It would be awesome to combine the two separate G force indicators to one that is a 2D graph, similar to the following one (attached) to the left.
[image]

jilleb, 2019-06-19:
I agree, that would be pretty awesome!
I've been looking for an easy way to do this, but so far I haven't been able to make it in a nice way.

xDefcon, 2019-06-20:
@jilleb The concept is a 2d graph with an image (a dot) that moves on it. You could take a look here: https://github.com/karamsa/android-radar
https://github.com/smasini/AndroidRadar

Starting from that you should be able to achieve the specified g-meter.

Hope this helps.

gennematte, 2021-06-24:
After 2 years, confirm that would be great!

jilleb, 2021-06-24:
First, it has to be stable for all 😏