jilleb/mqb-pm #47: The application restarts constantly

jdzuri, 2019-06-11:
The application restarts constantly and is not usable (it starts, it stops and it is restarted, like that all the time).

I used the default configuration, as it comes after installing it to prevent the error was for any specific option.

Environment info:

- Car: Seat Leon MY18 
- Display: MIB2 
- Mobile: Xiaomi MI5
- OS: Lineageos 16 (root)
- AA: version 4.3.591854

jilleb, 2019-06-19:
What firmware version is on your MIB2? (hold MENU)

jdzuri, 2019-06-19:
Hello, I'm not really sure about the MIB2 but I think so (it has only two buttons on the left, one for the multimedia and one for the full link).

- Device number: 5QA035824B
- HW: H41
- SW: 0468
- Codec media: 4101.0.0.0

jilleb, 2019-06-19:
Allright, MIB2 STD with software version 0468. Your car is sending data, and the app is receiving it. Make sure you've turned off any taskkillers, memory managers or battery savers that might kill MQB-PM or the Vag service.

jdzuri, 2019-06-19:
Yes, I saw in the logs that the data was arriving but I did not know if there was an error in the parsing of them or some error not controlled.

The restart is constant, in a loop, it starts, it stays unresponsive and it restarts (in a few seconds, 4-5, the whole process).

In principle everything you say is correct, I have it out of the battery optimization, etc. I'll check it in case there's something else.

If it does not work with the settings that may have been pending of those that you comment, next week I will compile a modified version to try/catch more errors and added traces to try to determine the point of possible error.

divacky, 2019-06-19:
I use aLogcat to debug. Enter mqb into the filter and filter out errors. Here we see an error on a particular line
[image]

mobrembski, 2019-07-23:
I have expected a simillar issue.
Moreover, i have same headunit (MIB2STD), same brand and same model year. Also, same phone brand :) (expect that i have older Redmi Note 4 with original MIUI)
But i don't think this is an issue of firmware/headunit.
I've profiled & debugged app and found out that usage of main thread is very very intensive.
Almost 99% of time app spend on onMeasure for ConstraintLayout, and many times logcat shows up an info that onMeasure is very slow due to high main thread usage.
Don't know why, but i've tried to "lower" quality of graphics (change background, change display items etc.) - no luck

mobrembski, 2019-07-24:
Here are logs:
`07-24 14:31:32.401 30530-30530/com.mqbcoding.stats I/Choreographer: Skipped 136 frames!  The application may be doing too much work on its main thread.
07-24 14:31:33.812 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =android.support.constraint.ConstraintLayout{dcf25f1 V.E...... ......ID 0,70-800,385 #7f0900c8 app:id/include}time =1409 ms
07-24 14:31:34.714 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =android.support.constraint.ConstraintLayout{dcf25f1 V.E...... ......ID 0,70-800,385 #7f0900c8 app:id/include}time =896 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =android.support.constraint.ConstraintLayout{b0efc4e V.E...... ......ID 0,0-800,400 #7f0900cf app:id/layoutDashboard}time =2313 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =android.widget.FrameLayout{4df9f7e V.E...... ......ID 0,0-800,400 #7f090093 app:id/fragment_container}time =2314 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =com.google.android.gearhead.appdecor.AppContentLayout{74e2c71 V.E...... ......ID 0,0-800,400 #7f110135 app:id/content_view_container}time =2314 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =android.widget.FrameLayout{7a2ff8a V.E...... ......ID 0,0-800,400}time =2314 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =com.google.android.apps.auto.components.drawer.MotionFilteringDrawerLayout{806242a VFE...... .F....ID 0,0-800,400 #7f110134 app:id/drawer_layout}time =2314 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =cwx{26b6ca0 V.E...... ......ID 0,0-800,400}time =2314 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =com.google.android.gms.car.ea{ffbab83 V.E...... ......ID 0,0-800,400}time =2314 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =android.widget.FrameLayout{fca2ecd V.E...... ......ID 0,0-800,400 #1020002 android:id/content}time =2314 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =android.widget.LinearLayout{64c95f7 V.E...... ......ID 0,0-800,400}time =2314 ms
07-24 14:31:34.715 30530-30530/com.mqbcoding.stats D/View: [ANR Warning]onMeasure time too long, this =com.android.internal.policy.PhoneWindow$DecorView{1cce31b V.E...... R.....ID 0,0-800,400}time =2314 ms
07-24 14:31:34.759 30530-30530/com.mqbcoding.stats I/Choreographer: Skipped 142 frames!  The application may be doing too much work on its main thread.`

divacky2, 2019-08-19:
For me, the cause was always found in the log. Android fills the log with a lot of information and you need to filter the data or search.
The Android Auto rebooted itself on the previous phone when the phone was overloaded and was busy.

mobrembski, 2019-08-19:
If so, that means that only one we can do is to flattern/slim down app layouts.
App still spends almost 98% of its main thread time due to relayouting.

divacky, 2019-08-19:
Yes. We read all the data that the EXLAP service returns. We register all service items. It would only register items that are needed in the application.

divacky, 2019-08-19:
To connect to exlap:
1) connect
2) authentication
3) list of available items
4) registration of all items in the list (200 items for me)
5) exlap fills data into object Map
6) I read in the app 20-25 items

mobrembski, 2019-08-19:
I think you misunderstood me.
There is no problem with reading data from exlap itself. We can read all data at once with no bootleneck.
The problem is with android layouting. We extensively use ConstraintLayouts, and most of TextViews we have attribute wrap_content for width and height. This has huge impact on performance. Why?
I will give you an example on textview. When we call setText on textview containing wrap_content with value (already received from exlap), then android will force doRelayout on whole screen.
This means that all views must recalculate its sizes to get correct layouting. And it's done FOR EACH call for setText. This cost a lot of performance.
I only analyzed it for textview, not sure what about SpeedView lib, maybe it also calls doRelayout when value changes on gauge... i don't know

For ex. i have 10 cores Xiaomi Redmi Note 4 with android 6.0. By default, this app is not usable - i can get maybe 1 FPS of screen refresh? maybe less.
When i changed all textviews to fixed width and height, i got about 20 FPS of refresh rate. Now it's usable, but i belive there is still a room for imprevement.

But setting a fixed width and height has disadvantage - some views may be incorrectly show on screen.
Currently i have no idea how to speed up this app...

jilleb, 2019-08-19:
I think the textviews are the way to go, but...setting it to 100dp for all textviews will break the layout of the dashboard view. We should look into this, how to resize while still keeping everything in the right place.

jilleb, 2019-08-19:
Okay, I have some ideas.

- I'll look into the text views and static size
- elements that are hidden don't need updates. For instance, if high visibility rays are off, it is hidden, but still gets updated. Same goes for min max clocks. I think we can save some resources there.
- smaller images. For some clock backgrounds, I could improve the graphics by making the graphic smaller. Should save some memory? 
- make updates frequency variable through settings. It currently updates 4x per second, maybe some don't need such a frequency.

Any thoughts?

jilleb, 2019-08-20:
Made text elements have a static width, with commit 77d03bb9723b2b31ba7f4920f5731e5d99e60044.

mobrembski, 2019-08-20:
I've prepared a small pull request.
I'm quite bit surprised but i replaced match_parent into 0dp (it's not recommended to use match_parent inside constraintLayout), and simplified fragment_dashboard, and now app is more responsive, CPU load dropped by few percents and i didnt see a much ANR measure Too long messages in logcat.
There is still a room for improvement, (for ex. menu animation is a little big laggy on my phone), but definitely it's still better than it was.
https://github.com/jilleb/mqb-pm/pull/74

jilleb, 2019-08-20:
Great work. I've merged them. Will test tonight.
I will look into not-updating elements that are hidden. Not sure if this is really having any impact, but we can try :-)

jilleb, 2019-08-20:
@mobrembski , I am testing it. I've noticed some strange behavior, especially with the displaying of graphs and rays. I'll see what i can do to fix.

mobrembski, 2019-08-20:
Hmm, i havent see any problems with rays. But indeed, i see a problem with graphs. I will take look at it as well.

jilleb, 2019-08-20:
The rays are shifted slightly from the center. I made the sizes fixed instead of 0, that fixes it. Let's test it :-)

jilleb, 2019-11-14:
I think it's fixed...

hllhll, 2019-11-21:
Here i'm having similar issue,
App opens (AA3 with enabler) does needle sweep on 3 gauges, crashes and restartx3. Is that the same issue?

(Seat Arona 2018 MIB2 STD+NAV will add software ver later)

Data is received by the app, since I can see the oil temperature notification when it reaches 70deg when app is in background.

jilleb, 2020-01-18:
Fixed in the latest versions. If it still happens, remove all data from the app and try again.

closed 2020-01-18 by jilleb