jilleb/mqb-pm #48: Forward data to exlap/canbus for displaying in Cluster

jilleb, 2019-06-19:
@divacky, I've been looking in this for a long time, maybe you can assist..
Currently we are reading out data on the exlap channel, but we can also post data to it. I wonder if it's possible to get data back into the car, for instance for displaying in the cluster, like Polar FIS does:

[image]
[image]

divacky, 2019-06-19:
POLAR FIS and similar devices do not send any data to the system just displaying audio information. The name of the song displays the values from the connected connector to the MQB socket. It's HW. I have tried different options and can eg control the radio or search for songs. Next, I tried to connect with a web service that would allow more.

jilleb, 2019-06-19:
I believe they modify the music-information on the canbus and insert their own data instead which is then interpreted by the Cluster.

jilleb, 2019-07-30:
With the  following code, we should be able to put data to the exlap channel:

```
CarVendorExtensionManagerLoader vexLoader =
        (CarVendorExtensionManagerLoader)mCar.getCarManager(
                CarVendorExtensionManagerLoader.VENDOR_EXTENSION_LOADER_SERVICE);
mVexManager = vexLoader.getManager(getVendorChannelName());
if (mVexManager == null) {
    throw new RuntimeException("Vendor channel not available");
}
mVexManager.registerListener(new CarVendorExtensionManager.CarVendorExtensionListener() {
    @Override
    public void onData(CarVendorExtensionManager carVendorExtensionManager, byte[] bytes) {
        // Do something with data
    }
});
mVexManager.sendData(new byte[] { (byte) 1, (byte) 2, (byte) 3 });
```

jilleb, 2021-06-21:
Almost there, but that's for a different repository ;-)

closed 2021-06-21 by jilleb