jilleb/mqb-pm #59: Version number is missing in the app-list

jilleb, 2019-07-30:
[image]

mobrembski, 2019-08-05:
Unfortunately, i cannot reproduce it since new android auto design is not available in my country :/
(I've updated Android Auto to 4.5, latest version but i still have old layout in car).
However, i've found out that Application Manager in android settings show up that Performance Monitor has version number null :)
This patch fixes it, and i belive it will also fix this one
https://github.com/jilleb/mqb-pm/pull/63

jilleb, 2019-08-05:
Haha thanks! I'll merge it soon :-)

jilleb, 2019-08-11:
Merged with caa59b856027f18d8f1b9e91741303111d1d5fd1, closing. Thanks!!

closed 2019-08-11 by jilleb