jilleb/mqb-pm #60: [FATAL] Buildbreak after upgrade to Androidx

mobrembski, 2019-07-31:
Hi all,

I've just upgraded to current master, and on my phone app stopped to run, due to:

`    android.view.InflateException: Binary XML file line #11: Binary XML file line #447: Error inflating class <unknown>
        at android.view.LayoutInflater.inflate(LayoutInflater.java:543)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:427)
        at com.mqbcoding.stats.DashboardFragment.onCreateView(DashboardFragment.java:295)
        at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2595)
        at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:881)
        at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1238)
        at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:434)
        at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2076)
        at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1866)
        at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1821)
        at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1727)
        at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2660)
        at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2610)
        at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:246)
        at com.google.android.gms.car.e.onStart(Unknown Source)
        at com.google.android.apps.auto.sdk.CarActivity.onStart(Unknown Source)
        at com.mqbcoding.stats.MainCarActivity.onStart(MainCarActivity.java:249)
        at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:550)
        at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:224)
        at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:746)
        at com.google.android.gms.car.internal.h.run(CarActivityHostImpl.java:2)
        at android.os.Handler.handleCallback(Handler.java:815)
        at android.os.Handler.dispatchMessage(Handler.java:104)
        at com.google.android.gms.c.a.a.b.dispatchMessage(TracingHandler.java:9)
        at android.os.Looper.loop(Looper.java:207)
        at android.app.ActivityThread.main(ActivityThread.java:5845)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:907)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:768)
     Caused by: android.view.InflateException: Binary XML file line #447: Error inflating class <unknown>
        at android.view.LayoutInflater.createView(LayoutInflater.java:649)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:768)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:708)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:839)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:802)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:842)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:802)
        at android.view.LayoutInflater.parseInclude(LayoutInflater.java:975)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:835)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:802)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:519)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:427) 
        at com.mqbcoding.stats.DashboardFragment.onCreateView(DashboardFragment.java:295) 
        at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2595) 
        at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:881) 
        at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1238) 
        at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:434) 
        at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2076) 
        at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1866) 
        at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1821) 
        at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1727) 
        at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2660) 
        at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2610) 
        at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:246) 
        at com.google.android.gms.car.e.onStart(Unknown Source) 
        at com.google.android.apps.auto.sdk.CarActivity.onStart(Unknown Source) 
        at com.mqbcoding.stats.MainCarActivity.onStart(MainCarActivity.java:249) 
        at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:550) 
        at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:224) 
        at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:746) 
        at com.google.android.gms.car.internal.h.run(CarActivityHostImpl.java:2) 
        at android.os.Handler.handleCallback(Handler.java:815) 
        at android.os.Handler.dispatchMessage(Handler.java:104) 
        at com.google.android.gms.c.a.a.b.dispatchMessage(TracingHandler.java:9) 
        at android.os.Looper.loop(Looper.java:207) 
[... 80 lines cut ...]
        at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:550) 
        at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:224) 
        at com.google.android.gms.car.internal.CarActivityHostImpl.a(CarActivityHostImpl.java:746) 
        at com.google.android.gms.car.internal.h.run(CarActivityHostImpl.java:2) 
        at android.os.Handler.handleCallback(Handler.java:815) 
        at android.os.Handler.dispatchMessage(Handler.java:104) 
        at com.google.android.gms.c.a.a.b.dispatchMessage(TracingHandler.java:9) 
        at android.os.Looper.loop(Looper.java:207) 
        at android.app.ActivityThread.main(ActivityThread.java:5845) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:907) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:768) `

Most interesting thing here is it:
`     Caused by: android.content.res.Resources$NotFoundException: Resource ID #0x7f0e0055
        at android.content.res.Resources.getValue(Resources.java:1561)
        at android.content.res.MiuiResources.getValue(MiuiResources.java:147)
        at android.content.res.Resources.getColor(Resources.java:1112)
        at android.content.res.Resources.getColor(Resources.java:1079)
        at androidx.cardview.widget.CardView.<init>(SourceFile:19)
        at androidx.cardview.widget.CardView.<init>(SourceFile:3)`

I've tried to remove  app:cardBackgroundColor paramter from layout_dashboard_gauges.xml, but it won't help.

My Phone is Android 5.1 based, maybe androidx is not working correctly on such API level?

jilleb, 2019-07-31:
😭 that sucks.. 
I merged your PR, thanks 👍

jilleb, 2019-07-31:
@mobrembski , now I'm having issues building, getting errors related to ids.xml:

`Cause 1: java.util.concurrent.ExecutionException: com.android.builder.internal.aapt.v2.Aapt2Exception: Android resource compilation failed
C:\Users\jille\.gradle\caches\transforms-2\files-2.1\f436ab1b3cb59c49cbaf9b88bf151cd1\res\values\ids.xml:20:3-22:10: AAPT: error: <item> inner element must either be a resource reference or empty.
    
C:\Users\jille\.gradle\caches\transforms-2\files-2.1\f436ab1b3cb59c49cbaf9b88bf151cd1\res\values\ids.xml:35:3-37:10: AAPT: error: <item> inner element must either be a resource reference or empty.
    
C:\Users\jille\.gradle\caches\transforms-2\files-2.1\f436ab1b3cb59c49cbaf9b88bf151cd1\res\values\ids.xml:2:3-4:10: AAPT: error: <item> inner element must either be a resource reference or empty.
`

mobrembski, 2019-07-31:
> AAPT: error: inner element must either be a resource reference or empty.

I think you need to clean and fully rebuild project.
We don't use ids.xml file

jilleb, 2019-07-31:
Ahhh that works :-)
Thanks. I'll commit some small changes.

closed 2021-05-10 by jilleb