jilleb/mqb-pm #72: Cross-check all min/max and resolutions of data elements with schema.json

jilleb, 2019-08-15:
Currently, most of the min/max values are based on personal preference or measurements. It would be nice if the min/max of a clock are taken from the schema upon creating or updating the clocks.

mobrembski, 2019-08-19:
I disagree.
I'm not sure if shema.json is properly filled by car firmware. probably not, for ex. for my car it shows that maximum torque is for -1571 up to 1571 Nm.
Total mess, since my car has only about 100 Nm max :)
But i  would rather want to see an option to get default values from schema, and give user ability to override it from settings.

jilleb, 2019-08-19:
haha yeah fair point. But getting the data resolution/format/unit from the schema would be an improvent, I think.

closed 2020-01-10 by jilleb