jilleb/mqb-pm #79: Needle delay

davidmh78, 2019-11-06:
Hello, I understand that some delay is unavoidable due to the time needed to send the data, process it by the app and draw the screen, but sometimes it seems that the app gets the data some time before drawing the needle, in fact, it appears to draw the maximum value before the needle reaches it. Maybe the app takes some time to make the "transition" to new value (aka the needle animation) possible. 

Would it be possible to have some kind of immediate mode?

jilleb, 2019-11-06:
Yes, that's possible. Currently the clocks are animated to transition between each value in a smooth fashion. I tested a functionality where it was turned off, and it wasn't as good looking, but accurate. 

I'll add a switch to the settings: accurate display/nice display 😁

Note: the marks for max and min values are not animated, you can see them going very fast

davidmh78, 2019-11-06:
Sounds great! Yes I noticed max and min moved faster, thats how I realized there was an animation delay :)

jilleb, 2019-11-12:
Fixed it in 34d582ea3afebbb76104c900fcec077f644ca6a0 :-)
Happy testing!

closed 2019-11-12 by jilleb