jilleb/mqb-pm #82: Not working?

Leproide, 2020-01-06:
Hi!
I want to test this app with an old rooted Moto G with:
CyanogenMod 14 (Android 7.1.1)
Performance Monitor 0.7B
Android Auto 4.9.594933
Phenotype Patcher 0.85 (For enabling the performance monitor on AAuto)
Google Play Services 19.8.31
VAG Extensions for android auto 2.4

And the 5F on my Composition Media (touch version) are:
Hardware Number: 3Q0035824C
Software number: 0842

The problem is:
-Without phenotype Android Auto not show me the performance monitor icon
-If i enable the performance monitor app with phenotype in any case the app wont start

Any suggestion?

jilleb, 2020-01-06:
Ahh sad to hear that. Can you supply a logcat?

Leproide, 2020-01-06:
[logcat_01-06-2020_20-01-07.txt](https://github.com/jilleb/mqb-pm/files/4027433/logcat_01-06-2020_20-01-07.txt)

Hi jilleb, thank you very much.
This is my logcat.
I tried to reinstall, reset permission etc but dont work, open and close in 1ms with black screen.

Leproide, 2020-01-15:
Hi :D
Some news about this problem?

jilleb, 2020-01-18:
Found nothing so far.

Leproide, 2020-01-18:
Today i update the 5F module, not working in any case :(
I dont have the HIGH mib, u think is this the problem?

jilleb, 2020-03-03:
That should not be a problem. One of my friends also tried to get it working with the latest CyanogenMod, but wasn't succesful. I can't test on all OS variants, so it's a hit-or-miss type of situation. It's plausible that it has something to do with google play services or something.

closed 2021-05-07 by jilleb