jilleb/mqb-pm #84: Oil and transmission tempature via Torque Pro

Step-han, 2020-01-12:
Hello,

Is it possible to get the option for oil and transmission tempature via Torque Pro? I'm using this app with a BMW F20 m135i and cannot show the oil/transmission now.

Greetings,
Stephan

[image]

jilleb, 2020-01-12:
Yes, that's possible. I included a couple of Torque transmission and oil temperatures, but I'm not sure if they are compatible with your car. Can you run look up the right PID, by using the Torque Scan plugin?

You will get a list like this, and I need the value between brackets (like [ff124f] )
[image]

Also: nice car! Looks like I need to incorporate a theme like this ;-):
[image]

Step-han, 2020-01-12:
I've made same screenshots from my phone with the plugin installed.

Soo it would be [5c] for engine oil temperature and [fe1805] for transmission temperature?

[image]
[image]

Yea the car is amazing, but the infotaiment of BMW sucks, now with a separate module I can have AndroidAuto on the widescreen, now the screen makes sense :)

Theme for BMW would be great!

Thanks for youre quick reply.

jilleb, 2020-01-13:
Thanks for the screenshots! I'll include these. I'm surprised I didn't include them before, especially since all kinds of exotic PIDs are there (timing advance and such). :-)

ps: dankjewel! :-)

Step-han, 2020-01-13:
Nice the hear! When u add those PIDs how can I test them? Do you release a new version of the app or can you send me a beta .apk?

jilleb, 2020-01-13:
I'll send you a beta apk :-)

Step-han, 2020-01-14:
Sounds promising :)

jilleb, 2020-01-14:
In the mean time.. I'll draw up some BMW theme ;-) 

[image]

inspiration:
[image]

Step-han, 2020-01-15:
Looks nice, even for a 15 min playtime in PS.

I saw some commits from yesterday that you adding the new torque PIDS. Does this mean I can test it with a beta .apk? New to github soo don't know how that works.

jilleb, 2020-01-15:
Almost. I need to do a test drive in my car and then I'll send you a beta.

jilleb, 2020-01-18:
@Step-han check it out ;-)

[Performance Monitor 0.88a.zip](https://github.com/jilleb/mqb-pm/files/4081383/Performance.Monitor.0.88a.zip)

jilleb, 2020-01-18:
Hmm, maybe I should try to go for this look:
[image]

What do you think?

Step-han, 2020-01-18:
I will test it tonight and make some pictures from my original system for inspiration!

Step-han, 2020-01-18:
I've done a testride. Some points:

- the elements between the clocks don't work anymore
- turbo boost pressure is in bar even if psi is selected. Switching it when the app is on AndroidAuto wil get psi view working but only to the max of 5 psi (my car is doing around 16-17 psi).
- and on the elementsview between the clocks I can't select transmission tempature, only on clocks. Is it possible to select it there?

Here some screens/photos:

[image]

[image]

[image]

My stock clocks from BMW:
[image]

Screenshot of the settings in the app:
[image]

And thanks for the great work! Oil and transmission temps are working fine in the clock view!

jilleb, 2020-01-19:
:-) 

yeah, I made some modifications to the torque text-elements and I wasn't 100% sure why I coded them like I did before... so I guess that broke them 🤣 

Thanks for the review! I'll fix them and get back with a new version. 
To be honest, I wish I could scrap the entire unit-support (metric vs emprical) , because it's making the app more complicated than it needs to be 😆

Step-han, 2020-01-19:
PSI is more accurate then BAR to read off for the turbo boost. Wider range. In the car performance world they always talk in PSI.

Maybe a view of PSI from -15 to +30

I did a drive today with the view of photo 2, coolant, oil, transmission and its nice to see the temperatures. :)

Step-han, 2020-01-25:
I saw some nice changes like a BMW theme :)

Keep up the good work and many thanks!

jilleb, 2020-01-26:
:-) yes.. some more test drives tomorrow, and then I'll release. I chose to let Torque decide what unit and min/max range should be used. Hopefully all will work as planned 😁

Step-han, 2020-01-31:
And how was the testdrive? 😁

Step-han, 2020-02-12:
Any chance to send me a new beta .apk?

Step-han, 2020-03-21:
I see you added a new public version, I'll will test it today. Will make soms photo's en screenshots for u.

Step-han, 2020-03-21:
I did a testrun. The Performance Monitor works great!

Only thing is when the performance monitor starts the turbo boost pressure is always in bar even if you have selected PSI. Switch to bar and back to PSI fix it for now. The range of PSI i 0-100 better is -15 to +30 for PSI.

Here some pictures:

[image]
[image]

sal989, 2021-07-15:
Reviving a old post, any chance this same oil and transmission readings can be done for a 135i n54? So far i have had no luck.
TIA

sal989, 2021-07-15:
[image]