jilleb/mqb-pm #86: Get units and min/max from schema for exlap elements.

jilleb, 2020-01-26:
Currently, the handling of units isn't done nicely by the app. There are a couple of challenges:

- users can choose units in the car-setup of their infotainment units. Exlap reports data in this unit. 
- users can choose units in the Torque-bhp app. 

I am thinking about removing all the unit overrides, to make unit-handling as clean as possible.
People sent me several requests like:
- the need for unit selection for each clock
- min max range selection for each clock.

But that would mean 3 additional controls for each of the 3x3 clocks (so 27 additional setup entries).

If I add an override, I need to make a unit converter class, that I can use to convert pressures, speeds, horsepower, etc.

something like this: 

```if (car-pressureunits = psi
and user-preference = bar) {
{
value = value/14.503773779685869;
} else if (car-pressureunits = bar
and user-preference = psi) {
{
value = value * 14.503773779685869;
} else { #in case preference and car unit is identical)
#do nothing
}

```

MisterNo1, 2020-05-04:
Hello,
At "Tank level" when I set "Fuel tank size" to 50l, my clock still has scale from 0 to 100. Because of that, when I bought the full tank, the tick value is on the half of the clock :). Can I change this option? Best regards

MisterNo1, 2020-05-04:
[image]