jilleb/mqb-pm #88: No connection

adrienturpin974, 2020-02-12:
Hi 
Performance Monitor work works perfectly when the car is started for the first time, but if I lose the connection with the phone (cable disconnect) and restart, I have no information going back up, I am obliged turn off the car and restart everything.

jilleb, 2020-03-03:
Can you try again with the latest version?

adrienturpin974, 2020-03-03:
With your 0.88 this issue, it's not resolved.

jilleb, 2020-03-03:
Make sure the app or vex vag service is not being killed by some battery saving measure. I've connected and disconnected many times during tests with various phones, no issues reconnecting. However, when some app killer like Sony Stamina mode is on and they are not on the whitelist, it's being a problem.

adrienturpin974, 2020-03-04:
Hi, in my 2 apps i've not restriction. 
On XDA another menber have a same issue on Xiaomi mi9

phoinixgrr, 2020-03-22:
I can confirm. The same here with Xiaomi Mi9. 
All relevant power saving features are disabled.

adrienturpin974, 2020-04-04:
> I can confirm. The same here with Xiaomi Mi9.
> All relevant power saving features are disabled.

I test your apk, have you found a solution to solve this problem?

JHanzz, 2020-05-03:
I can confirm, data stops showing when phone usb cable disconnect and reconnect back with v0.88B.
I do not have this problem in v0.7B.
Using rooted lineage and Seat.

JHanzz, 2020-05-17:
> I can confirm, data stops showing when phone usb cable disconnect and reconnect back with v0.88B.
> I do not have this problem in v0.7B.
> Using rooted lineage and Seat. VEX+performance monitor in no optimization battery mode.

Uninstalling aauto-vex-vag-2.4 and installing it again helps and performance monitor shows data if phone disconnect and reconnect back.