jilleb/mqb-pm #90: Torque app

iridium-x, 2020-02-27:
I realise. If I have torque app installed, regardless of if i have torque data elements selected, the PM app will not launch. most probably a crash. tried uninstalling torque. PM launches fine.

when i have torque installed, i have to open the torque app before launch PM in AA. 
I dont remember this was the case when i was using 0.7B late last year.

I recently reinstalled again, and have the above issue.

jilleb, 2020-03-03:
Good question.. can you try it with the latest version?

iridium-x, 2020-03-03:
I just tried the latest version. same issue. I had to manual open torque app. then launch PM in AA.
also when i disconnect my phone from AA, i have been manually closing the torque app. or it just drains my battery i no no time. lol..

this launching torque prior to launch PM in AA has got me fooled for quite a while. i kept thinking AA as giving me issues. 

anyway thanks for the latest update! looks good.

jilleb, 2020-03-03:
Strange... It can take about a minute before Torque gets any data from the car after starting PM. Are you sure you were not getting any data at all?

iridium-x, 2020-03-04:
Hi! Torque gets data no issues.
The issue is if i do not open torque prior, PM will not even launch in my MIB2. I cannot even see the 3 clocks. what happens is the screen goes black, then it goes back to the AA menu.

So if i open Torque before launching PM on my MIB2, all works fine. 

its like PM is not 'opening' a connection to torque app. i have to manually do it by launching Torque on my phone first.

jilleb, 2020-03-04:
That's really strange... did you allow plugin access in Torque?
It's using the default ways to start Torque and open a connection, as documented by the Torque developer.. strange stuff 😆

iridium-x, 2020-03-04:
haha. yes, allow plugin full access is checked. 
Torque version is 1.10.114

does PM always want to launch torque by default regardless of the parameters selected?
I tried selecting basic parameters (non torque) in PM. the PM will also not start in AA, as long as i have torque app installed.

Uninstalling torque fixes the issue. PM launches with issues in my MIB2 unit.