jilleb/mqb-pm #95: 'Crash' on app start

karimovrinat, 2020-06-09:
Hi to all!
Cannot start app on Xiaomi Redmi K20, MIUI 12, ver 20.5.29, Android 10
When I start app, screen is blinking and nothing happens.
In memory manager application is running.
And MIUI returns error, seems like dont have Activity Permissions, but all permissions granted in MIUI settings. 
`java.lang.RuntimeException: Unable to start activity ComponentInfo{com.lbe.security.miui/com.android.packageinstaller.permission.ui.GrantPermissionsActivity}: android.content.res.Resources$NotFoundException: String resource ID #0x7f19003d
 at android.app.ActivityThread.performLaunchActivity(Unknown Source:621)
 at android.app.ActivityThread.handleLaunchActivity(Unknown Source:47)
 at android.app.servertransaction.LaunchActivityItem.execute(Unknown Source:63)
 at android.app.servertransaction.TransactionExecutor.executeCallbacks(Unknown Source:77)
 at android.app.servertransaction.TransactionExecutor.execute(Unknown Source:73)
 at android.app.ActivityThread$H.handleMessage(Unknown Source:43)
 at android.os.Handler.dispatchMessage(Unknown Source:19)
 at android.os.Looper.loop(Unknown Source:242)
 at android.app.ActivityThread.main(Unknown Source:98)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(Unknown Source:11)
 at com.android.internal.os.ZygoteInit.main(Unknown Source:275)
Caused by: android.content.res.Resources$NotFoundException: String resource ID #0x7f19003d
 at android.content.res.Resources.getText(Unknown Source:36)
 at android.content.res.MiuiResources.getText(Unknown Source:11)
 at android.widget.TextView.setText(Unknown Source:8)
 at com.android.packageinstaller.permission.ui.GrantPermissionsDefaultViewHandler.updateDescription(Unknown Source:215)
 at com.android.packageinstaller.permission.ui.GrantPermissionsDefaultViewHandler.updateAll(Unknown Source:0)
 at com.android.packageinstaller.permission.ui.GrantPermissionsDefaultViewHandler.updateUi(Unknown Source:27)
 at com.android.packageinstaller.permission.ui.GrantPermissionsActivity.showNextPermissionGroupGrantRequest(Unknown Source:549)
 at com.android.packageinstaller.permission.ui.GrantPermissionsActivity.onCreate(Unknown Source:605)
 at android.app.Activity.performCreate(Unknown Source:19)
 at android.app.Activity.performCreate(Unknown Source:1)
 at android.app.Instrumentation.callActivityOnCreate(Unknown Source:3)
 at android.app.ActivityThread.performLaunchActivity(Unknown Source:376)
 ... 11 more`

alexandro144, 2021-05-26:
Good morning
  also have such a problem
I have installed performance and Vag Extensions for android
I try to start the app, the scheme shows the impulse but does not start
I tried removed the application and reinstalled it but nothing changes
How can I solve the problem  ?

gennematte, 2021-06-18:
After gave all permission to app, have you put "power" indicator on screen?
For me same problem, removing it was the solution

jilleb, 2021-06-18:
I'll see where Power goes wrong.. haven't had any issues it with it myself, but removing it for the time being is fine.

Salcedano, 2021-11-16:
I have a problem and it is that the app does not appear on the AA screen.
The "performance monitor" settings icon does appear, but when I access it, the text appears: No messages were received while driving.
My mobile is a REALME 6 pro
Is there any solution?.
Thanks and Thanks for the work

Merkta, 2021-11-18:
> Tengo un problema y es que la aplicación no aparece en la pantalla AA. Aparece el icono de configuración del "monitor de rendimiento", pero cuando accedo a él, aparece el texto: No se recibieron mensajes mientras conducía. Mi móvil es un REALME 6 pro ¿Hay alguna solución ?. Gracias y gracias por el trabajo

Tengo un realme X2 pro y me sucede lo mismo...

Hay alguna solución? Se sabe de donde es el problema? 

Gracias por el magnífico trabajo mis respetos

jilleb, 2021-11-18:
Please write in English.