jilleb/mqb-pm #102: Support for Seat Leon 2020 FR

RealIndica, 2021-04-30:
Hi!
I am using an s21 rooted.
I have installed the latest vex-vag service and got the latest performance monitor.

It seems it is completely not working.
The dash does not update and the guages stay on 0.

Under raw data, it says there's no connection and no data.

Hardware : H56
Software : 1664

A fix would be highly appreciated.

MeJebus, 2021-05-06:
A fix would be awesome! I have a 2021 Skoda Kamiq with MIB 3. I have a sneaking suspicion that the MIB version is the problem. In the description it says compatible with MIB2 Std, MIB2 High and MIB2,5.

I would love to know when it will come available for us!

jilleb, 2021-05-06:
Using the app on my own 2020 Leon FR... But depending on the radio type it will not work. I can't add support for radios that don't have any data on the internal exlap channel. So.. I can't add support for a specific car or radio type.

MeJebus, 2021-05-06:
What radio is in your 2020 Leon? MIB3? Maybe i'll get an ODB2 adapter and use it with Torque if my radio does not support exlap.

RealIndica, 2021-05-07:
> Using the app on my own 2020 Leon FR... But depending on the radio type it will not work. I can't add support for radios that don't have any data on the internal exlap channel. So.. I can't add support for a specific car or radio type.

What is the hardware and software version of your unit? And is there any other steps I can follow to make it work? Just to be clear I have the mk4.

Tonitoto11, 2021-05-07:
Buenas yo tengo Seat Leon FR 2020 mk4, no consigo hacerla funcionar.
Todo autorizado veo los indicadores pero no recogen i formación, pasos para hacerlo funcionar

jilleb, 2021-05-08:
> What radio is in your 2020 Leon? MIB3? Maybe i'll get an ODB2 adapter and use it with Torque if my radio does not support exlap.

I have an MIB2HIGH. MIB3 does not support exlap communication.

closed 2021-05-10 by jilleb