jilleb/mqb-pm #104: crash volkswagen tiguan 2017

alby160, 2021-05-04:
Buongiorno , ho una volkswagen tiguan 2017 con discover media (vesione software 0743),
vesione performance monitor v0.88B + aauto-vex-vag-2.4.

L'applicazione funziona , i dati vengono letti ma dopo qualche minuto mi manda in crash il discover con riavvio dello stesso.

ho provato cambiare le impostazioni , ma nulla.

cosa posso fare? c'è ina soluzione?
grazie

jilleb, 2021-05-10:
Please write in English.

closed 2021-05-10 by jilleb

fitojb, 2021-06-15:
Translation from Italian:

> Good day, I have a 2017 Volkswagen Tiguan with Discover Media (software version 0743), Performance Monitor v0.88B plus aauto-vex-vag-2.4.
The app works, the data are read, however after some minutes, Discover crashes on me and restarts.
I’ve tried to change settings, but no luck.
What can I do? Is there a solution?
Thanks