jilleb/mqb-pm #106: Problema performance monitor

pluto83s, 2021-05-07:
Performance monitor la clicco sembra che si voglia aprire, ma non si apre.
Dopo di che si blocca il discover e si resetta da solo
Tiguan 2017 discover media 2a gen
Samsung galaxy s9+

jilleb, 2021-05-07:
Please speak English

pluto83s, 2021-05-07:
Performance Monitor I click it looks like it wants to open, but it doesn't open.
 After that the discover hangs and resets itself
 Tiguan 2017 discover media 2nd gen
 Samsung galaxy s9 +

jilleb, 2021-05-07:
Make sure no app killers are running, battery saving is off for this app and all rights are given in the settings app.

pluto83s, 2021-05-07:
nothing to do, he doesn't want to open up.  yet to all consents.  it looks like it crashes

jilleb, 2021-05-08:
In that case, please generate a logcat so i can see what's going on.

pluto83s, 2021-05-08:
excuse the ignorance, what is LOGCAT?