jilleb/mqb-pm #107: App Crash if Torque not manually started

acejavelin, 2021-05-08:
Sorry, thought I saw a bug report for this earlier, but can't find it now.

2021 Hyundai factory head unit
AA 6.3.611324
PM 0.88.0.debug
Torque 1.10.120
Pixel 4a Android 11 April 5 2021 security patch 

After I figured out if I had an "invalid" Torque element defined the app would just restart several times and eventually crash, meaning one that doesn't appear to be supported by Hyundai, the app worked great, but I had to start Torque manually the initial time. 

Also if switching to another app, cannot switch back to PM and subsequently finding it in the AA menu and launching it causes it to crash. If I manually open Torque on the phone then launch PM, it works as expected. 

Don't know about this either (may not be relevant), but PM will never appear in the common/frequent apps line of AA app menu

LiamvanLeusden, 2021-05-15:
Have you tried disabling battery optimisation for Torque? I had a similar issue that Torque would not automatically start and PM would crash anytime it was opened. But disabling battery optimisation fixed it for me (android 9)

Pedro-vk, 2021-05-26:
In my case I've excluded Torque from battery optimisation but still crashing if not open