jilleb/mqb-pm #108: What do I need to make performance monitor working?

Cricetazio, 2021-05-08:
Hi, a part from the app and vax.etc etc installed, do I need something special in my car? Like additional optional package or similar to make PM working? I own a Passat 2021 but no data at all. Thank in advance for your kind reply.
Andrea

jilleb, 2021-05-08:
Good morning

Your car needs to send data in the exlap channel of the infotainment system. Some infotainment systems do this, some don't. MIB2 High on recent firmware does this, but I'm not sure about others.

If you don't receive any exlap data , you can still use the app, with and OBD Dongle + Torque.

Cricetazio, 2021-05-08:
I see...so I guess my car has nothing of what you said above, like mib2 or any other gadgets like odb dongle and torque. Car is just a business version of the Passat with some optional like travel assist and other stuff but I guess it's missing something to get the PM working

jilleb, 2021-05-10:
Allright, sad to hear that!

closed 2021-05-10 by jilleb