jilleb/mqb-pm #112: Crash afther 2 or 3 time

177Tenacia, 2021-05-09:
Hallo, I have a Samsung S10e not rooted and VW T-Roc 2018, OBD2 and Torque Pro.

The app works perfectly for 2 or 3 times ina day and then starts to crash on the phone.

Gps auth disabled, the app is not killed by battery saver apps.

Disabled proximity features.

I tried to use both data from VAG or data only from Torque but nothing, after two or three connection, in a couple of days, it start crushing.

If I delete all data from the app memory and setup again it starts work again and then so on.

jilleb, 2021-05-10:
I've noticed that using the app together with Torque or GPS can be causing issues... There have been some major updates there in the last commits.. I'm thinking about removing all the bloat from the app, like GPS support (because frankly.. we don't need it)

Disabling GPS stuff in the app should do the trick

177Tenacia, 2021-05-10:
[image]

177Tenacia, 2021-05-10:
What I should do to send you the debug log? I think is something from this menu.

I tried again to reinstall the appa and it works again very well. I think is something related to the animation