jilleb/mqb-pm #113: [REQUEST]

LiamvanLeusden, 2021-05-14:
First of all, thank you for this incredible app. Having tried OBDIIAA, this is way better. I have an old Mitsubishi (MUT-2 protocol) and a Pioneer SPH-DA230DAB. I am using Torque for the data.

I do have a request though. I have a Torque extension for some MUT specific PID's. It would be very cool if you could manually add your own PID's to get data from (and to edit their names).

I would happily help you with bug testing for this (and maybe other features), but I do not have a rooted phone (installed via AAAD).

Also new to creating issues and feature requests on GitHub. If there is anything I could do better, I would like to know!

P.S.
This is a small unrelated thing, but I have some formatting issues, and I cannot read the lower digital sensor displays. However when I enable proximity features, that makes the upper bar disappear, so I can see them. This is a good workaround but not ideal of course.

Greetings from the Netherlands!

Pedro-vk, 2021-05-21:
Other option is add more PIDs, like torque, power, GPS data...

gennematte, 2021-06-28:
I also think would be good with more PIDs, specially cause to me for example without torque I can't obtain turbo pressure and power data

sal989, 2021-07-15:
Can custom pid's be added for bmw n54 motors? I cant read engine oil temp and the transmission temprature any help appreciated👍