jilleb/mqb-pm #116: [BUG]Incorrect OIl and DSG Temp GTI MK7

alexbartGTI, 2021-06-01:
In which file can I edit that the oil temperature and DSG data are obtained from my custom Torque PIDs?
 In other words, specify the name of my custom PID in Torque to obtain the values ​​.. since in my GOLF GTI MK7 the default values ​​of Torque for oil temperature and dsg are incorrect.

Thank you

alexbartGTI, 2021-06-02:
Oil:

[image]
[image]
For DSG:
[image]
[image]

This is my custom PIDs in torque.

Thanks.

jilleb, 2021-06-21:
Currently, I have no way to include custom PIDs from Torque, but that might be a good idea for the future.

jilleb, 2021-06-21:
Working on it.. testing tomorrow.

On the main view it currently has no icon, but maybe I'll add an icon selector or something like that.
[image]

In the settings, it's as follows:
[image]

[image]

Would be nice if it's possible to add multiple custom PIDs....

alexbartGTI, 2021-06-21:
> Working on it.. testing tomorrow.
> 
> On the main view it currently has no icon, but maybe I'll add an icon selector or something like that.
> [image]
> 
> In the settings, it's as follows:
> [image]
> 
> [image]
> 
> Would be nice if it's possible to add multiple custom PIDs....

Oh yeahh!! Great! Thanks.
Another idea, it would be very good, that it could automatically obtain and import the custom torque pid.

thanks again for your great work.

alexbartGTI, 2021-08-15:
Hi, have you been able to try it? Could you share the apk with the changes made?
I would love to try it.
Thanks a lot!

SaulGoodman1337, 2021-08-30:
> Oil:
> 
> [image]
> [image]
> For DSG:
> [image]
> [image]
> 
> This is my custom PIDs in torque.
> 
> Thanks.

Information for golf 7 gtd oil temperature PID.
Use 2211be for PID. Is working ! If anyone need more pids... Ask. I am trying to get all pids with a can sniffer now.

Now we need custom pids in performance monitor!! 
Please @jilleb ... Give us your APK ! :)

jilleb, 2021-08-30:
:-) Soon!
I have holidays next week, then I have more time to work on things.

SaulGoodman1337, 2021-08-30:
> 
> 
> :-) Soon!
> I have holidays next week, then I have more time to work on things.

can't wait... sorry :D

SpursMadDave, 2021-09-01:
Be great to be able to add custom pids, I have DSG temp, Oil Temp and Haldex % all working in torque just need to be able to get them on performance monitor 👌

SaulGoodman1337, 2021-09-05:
> Be great to be able to add custom pids, I have DSG temp, Oil Temp and Haldex % all working in torque just need to be able to get them on performance monitor 👌

Yep we need this! 
Hopefully @jilleb get time this week

jilleb, 2021-09-05:
First the family needs me 😁

SaulGoodman1337, 2021-09-05:
> First the family needs me 😁

Haha yes understandable.
Make it better than my father ... :)

Currently I help myself with a magnetic holder for my phone. That also works ... But you know how it is with us technology crazy guys

SaulGoodman1337, 2021-09-11:
So your family had enough time. Now it's our turn. :D
No, do you already have an approximate schedule for us ?

jilleb, 2021-09-13:
Here's the schedule:

[image]

J/k, but seriously... It's done when it's done. I don't like to be rushed. And this is even more so on a project that's free of charge for everyone, and which exists because of my personal passion. I'm enjoying the first real vacation with my family in over 2 years, and then there's my job waiting for me soon again as well. Of you really depend on the custom Torque PIDs, compile the latest commit. It has an experimental custom Torque PID. But please...don't ask for a schedule because my free time and passion will not be scheduled.

aarons3, 2021-09-14:
I have a list of pretty much all MQB Simos18 PIDs.  PID, equations, etc.

Unfortunately my MIB firmware/version isn't supported, so I really haven't used this app much, but always thought it was really cool (wish it worked).  Happy to contribute/help where I can.

SpursMadDave, 2021-09-14:
aarons3 would you share your list?

I have a few

Oil Temp
DSG Temp
Haldex Pressure
Haldex %

Dave

SpursMadDave, 2021-09-14:
My mib2.5 unfortunately isn't supported unfortunately which I don't understand because I have the sports dials etc working so there must be some kind of connection giving data to the mib?

aarons3, 2021-09-14:
I have even more but these are many of the tuning essentials.  Equation format is for Racechrono app.

https://docs.google.com/spreadsheets/d/17nKlgeqHBmVnqw4OKaAGOJwZtXGkLNJo1yFMqe5oHck/edit?usp=drivesdk

SpursMadDave, 2021-09-14:
Thanks fir sharing, not sure how to convert these to work with Torque Pro though 🤔

SaulGoodman1337, 2021-09-14:
> Here's the schedule:
>
> [image]
>
> J/k, but seriously... It's done when it's done. I don't like to be rushed. And this is even more so on a project that's free of charge for everyone, and which exists [...]

Ok wow... actually I was just making a joke but if you mean to be pissed off right away.... here you go...

I can't find any commit to the custom pid here, neither in the master nor in the dev branch.

aarons3, 2021-09-15:
> Thanks fir sharing, not sure how to convert these to work with Torque Pro though 🤔

Sorry for the slow response.  Yes, just a little bit of formatting difference between Torque and Racechrono.

UBYTE is a 1 byte value
UWORD is a 2 byte unsigned value
SWORD is a 2 byte signed value

You have to write these in A, B, C variable format in torque pro which is a little wonky but here is a conversion cheat sheet:

UBYTE
x = A

UWORD
x = (A*256+B)

SWORD
x = (signed(A)*256+B)

Hope this helps!

SpursMadDave, 2021-10-11:
Got an Android Box plugged into my mib2.5 now so direct access to Torque Pro and all custom Pids 😁
Will keep checking back here just out of interest as its a nice neat way of integrating with AndroidAuto especially if custom PIDs can be sorted 👍