jilleb/mqb-pm #119: [BUG]

cptnuss-ops, 2021-06-07:
Since last major Android Auto-Update, the performance monitor isn't showing up anymore on the Display of the Car. Therefore the app can't be started, but on my mobile the status message "performance data will be logged" (might be actually a different message, but the meaning is the same ;)) is displayed even this explicit option is turned off in the app itself. I tried reinstalling the app with the VAG extension because the car beeing used is a Seat Leon ST FR, but without any luck. 

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S10+
 - OS: Android 11
 

**Additional context**
If I can assist you with any additional data you might need, you can't contact me via anytime. 

Thanks for your work an support in advance! 

Best regards, 

Lukas

gennematte, 2021-06-14:
Hi, same for me
Done every step, no battery saver, app on smartphone is ok, installed on AA, (also tried to remove all and re-install it) when AA start appear the notify on smartphone, but if I start it on car, it make a flash and close itself.
If you need everything, tell me.

Smartphone is an LG G7 with Android 10
Car is a Golf 7.5 with discover media

I add a CarLog (hope is the correct folder took from smartphone, tell me if it isn't)

*Additional Content*
For a time, just one (the first), it worked, from the second it stop.
Removed all and re-installed, doing the same things of before, but don't work

[CarLog](https://drive.google.com/folderview?id=1TrnloXRAC-F4FTkhuUIVSX39i5JJyCM2)

StephanBruh, 2021-06-14:
Known issue on android 11 + newest android auto. You can easily downgrade your android auto on your phone here to make it work again: https://github.com/shmykelsa/AAAD/wiki/How-to-downgrade-Android-Auto

gennematte, 2021-06-14:
Just returned to home and tried using 6.3 AA but still not work

StephanBruh, 2021-06-14:
> Just returned to home and tried using 6.3 AA but still not work

Hmm I had the same problem, I did install it via [AAAD](https://github.com/shmykelsa/AAAD) instead of direct via github maybe that will help for you?

gennematte, 2021-06-14:
I also use AAAD, bought the pro to made all this try

jilleb, 2021-06-14:
It seems as though this issue only occurs on people who installed it through AAAD...

I'm working on a new version, which will no longer need the vex service. That will probably fix all issues. But in the mean time.

Also: it's hard to support other install methods than the one I've tested, developed and been using for a long time myself...

gennematte, 2021-06-14:
@jilleb 
My phone is not rooted, is there an other method without using AAAD? I tried AA phenotype patcher but it say "no suitable method"

jilleb, 2021-06-14:
If you have a VAG car (Audi, Seat, Skoda, VW, etc), with a MIB2High infotainment system, you can use my mib2toolbox to patch the AA config on the Infotainment itself.

gennematte, 2021-06-14:
I have mib2 discover media (pretty sure it isn't high infotainment, if it is, tell me) 
But I have a big update, cause in all my try now I installed AA 6.2, and it work

gennematte, 2021-06-14:
Tried right now, returned to not work also with 6.2 AA...

gennematte, 2021-06-16:
Hi everyone.
Good update (also for @jilleb), after a big number of try I start using 6.2 (but I think version is not a problem, cause the error initially was the same of other) and at the first time it worked, than stopped however. The only change I made were the modify to the value on display so I modify 1 value for time on display (disconnecting and reconnecting AA everytime) and I noticed that the problem  was the power misurement (the normal, not hybrid power, cause my car is a GTE and hybrid power work perfectly) took from VW, not from torque.
That's it, hope to be helpful