jilleb/mqb-pm #121: [BUG] PROBLEMI CON PERFORMANCE MONITOR

Fabio1968, 2021-06-19:
Salve a tutti,
ho installato la versione di android auto 6.3.11323 su un xiaomi mi note 10 android 11 per una vw TCROSS.
Dopo aver installato da AAAD l'opzione PERFORMANCE MONITOR ed installo anche EXTENSIONS FOR VAG, quando inserisco
i dati che voglio visualizzare (per esempio la carica della batteria oppure il numero dei giri del motore), il sistema non mi visualizza nulla. Ho dato tutte le autorizzazioni ma non c'è verso di far funzionare gli indicatori.
Sapete darmi dei Consigli?
Grazie a tutti

jilleb, 2021-06-19:
Please write in English.

Fabio1968, 2021-06-19:
Hello everyone,
i installed android auto version 6.3.11323 on a xiaomi mi note 10 android 11 for a vw TCROSS.
After installing the PERFORMANCE MONITOR option from AAAD and I also install EXTENSIONS FOR VAG, when I enter
the data I want to view (for example the battery charge or the number of engine revolutions), the system does not show me anything. I have given all the permissions but there is no way to make the indicators work.
Can you give me some advice?
Thank you all

jilleb, 2021-06-19:
What infotainment system hardware and software do you have?

Fabio1968, 2021-06-19:
Dispositivo 2GM035842
HARDWARE 724
SOFTWARE 0876
Codec Media 1.2.0
Thank you

Fabio1968, 2021-06-19:
[image]

Fabio1968, 2021-06-20:
Can anyone help me understand the problem? thank you

jilleb, 2021-06-20:
This software version probably has no Exlap communication.
It's not in the list of softwares that are confirmed working.

Fabio1968, 2021-06-20:
Will software development be possible for this version as well? thank you

Scarica Outlook per Android<https://aka.ms/AAb9ysg>

jilleb, 2021-06-20:
If VW didn't enable the Exlap channel on that firmware, theres nothing I can do about it. There's no way to reach the data.

closed 2021-06-21 by jilleb