jilleb/mqb-pm #123: [BUG] Perfomance monitor crash everytime

Michelemk8, 2021-06-20:
hi, i wanted to know why in my sync ford fiesta mk8 the performance monitor app keeps closing indefinitely, i try to open it but then it closes immediately. I followed all the tips you wrote, android auto in developer mode and debug mode active from the phone settings. How do I resolve?

Samsung galaxy s20
Android 11
Android auto (downgrade) v6.3.6.11324 
The latest version of android auto not work.

Fiesta mk8 sync2.5

jilleb, 2021-06-20:
Can you make a debug log and send it to me?
Also, what data elements are currently selected to show?

Michelemk8, 2021-06-20:
[image]
It is full of .log files. How can I show you all of them in detail? the first page where I made the screen (which results in the last logs) are all the same written apart from a little bit in the timetable, I send you some images

jilleb, 2021-06-20:
Going by this report: Disable GPS functions of the app, that should do the trick.

Michelemk8, 2021-06-20:
ok i try, but i deactivate the gps of the phone or the permissions of the gps that the app asks?

Michelemk8, 2021-06-20:
[image]
do you mean that option there? I had it activated, now I have deactivated it. Or do I have to disable something else?

jilleb, 2021-06-20:
Disable the street name in title bar as well. I have to look into the code. This specific part wasn't coded by myself, but as of lately a lot of issues popped up around it..

Michelemk8, 2021-06-20:
That? [image]

jilleb, 2021-06-20:
I don't speek Italian 😏

No, I mean the setting "Show street name on Title bar" in the Performance Monitor settings

Michelemk8, 2021-06-20:
that is the screen where the android asks for the geolocation permission, I put always refuse

jilleb, 2021-06-21:
Try disabling the setting "Show street name on Title bar" in the Performance Monitor settings

Michelemk8, 2021-06-21:
Done... crash everytime

gennematte, 2021-06-21:
I intrude Just cause it seems a common problem, have you insert the "power" indicator on screen?
Try to run the app only giving it permission for all and removing "street name on title bar" (like jilleb say), don't modify display information

PlasmaPod, 2021-07-09:
I had this issue and this is what fixed it for me on my
Galaxy Note20 Ultra 

1. Settings
2. Apps
3. Find 'Performance Monitor'
4. Scroll down in the app info
5. Tap 'Battery'
6. Tap 'Optimise Battery use'
7. Where it's says 'Apps not optimised' tap the arrow beside it
8. Tap 'All'
9. Find 'Performance Monitor' and switch it off
10. Find 'VAG Extensions for Android' and switch that off as well

Hopefully should work after that. I think the app needs to run a service in the background to open when you select it on your headunit

Other steps
• Allow the app all its app permissions in app info