jilleb/mqb-pm #124: [BUG] Installation steps update

mobrembski, 2021-06-21:
Hi all,

I'm a big fan of mqb-pm, and also an author a bunch of commits in this project.
I've left project one year ago, and yesterday i wanted to come back. But problem is, that i cannot force android auto to run this app.
My configuration is as follow:
1. Rooted phone with AOSP firmware
2. Installed android auto from apkpure (version 6.6)
3. Updated Play services from OTA
4. Installed newest mqb-pm (builded from git) (using adb shell pm install -i "com.android.vending")
5. Installed aauto-vex-vag as it was in instructions (using adb shell pm install -i "com.android.vending")
6. Patched phenotype using AA-Phenotype-Patcher app.
7. Turned on developer mode and checked unknown sources in Android Auto settings.

Still Android auto doesn't recognize mqb-pm.
I think something is missing, or there is a specific version of AAuto required.
But if so, we need to update README :(

jilleb, 2021-06-21:
Hi,

You can try the AAAD app:
https://www.youtube.com/watch?v=lSxwTJhjM78

mobrembski, 2021-06-21:
I will try, thanks :)

But good question, even if it works how i can debug app  for development?

jilleb, 2021-06-21:
Good question :-) 

do you have a MIB2High infotainment system? If so, I've made a patch that can enable all custom apps on AA by changing a config file on the unit.

I'm currently trying to get my development environment for MQB-PM up and running again.. some gradle issues and I'm a bit rusty. I mostly test with Desktop Head Unit when I'm not in the car.

mobrembski, 2021-06-21:
Unfortunately, no :( i have MIB2Std+ from LG.

Interesting, looks like now its quite hard to develop ?

mobrembski, 2021-06-21:
Unfortunately, no luck. I've installed 6.3.611334 version of android auto, and used AAAD - and indeed, it worked :)
By work, i mean i can open performance monitor on car radio screen.

But problem is that it didn't retrieve any data from car. Tried to reinstall aauto-vex-vag, didn't help. 

I will try in few days a rooted phone, maybe it will work better.