jilleb/mqb-pm #128: [REQUEST] N54 Bmw monitors

sal989, 2021-07-17:
Hi, i am using the app on a android auto mmi box with my bmw 135i n54 motor and it has some bugs such as no engine oil temp reading and no boost reading. I know this is for VAG group but can these features please be implemented.

Want to be able to:
.have N54 motor PID'S for torque
.read the engine oil and transmission temprature
.have gearbox and boost displays working
.bonus have different displays and menus like the VAG group vehicles for the bmw group vehicles.

It can be done via apps like B-Tool expert and Bimmerlink but they do not work with AA it would be great to have the performance monitor work with the N54 motor

Many thank
A+++

sal989, 2021-07-17:
[image]

sal989, 2021-07-17:
[image]

nuwaus, 2021-10-06:
Can you please explain how did you get this working?
I'm trying to get it working on my BMW E90 as well, I have the MMI box and AA auto.
I'm trying to get Torque Pro data in it.

sal989, 2021-10-06:
Sorry I did not get any reply from anyone so it still does not work 😕