jilleb/mqb-pm #131: [REQUEST]: could the data on clocks be even quicker (less delay)?

EA888DLBA, 2021-08-07:
I'm monitoring torque and current fuel consumption (EXLAP) on clocks to check what ACC is doing (I'm annoyed by ACCs Downhill-braking and shut ACC off when consumption is 0)
I've noticed that even the numbers in the clocks are not as quick as in your text elements (/consumption dashboard) or on the car's MFA. In the settings app I've activated the Accurate clocks checkbox (comment says: Accurate values on clock).

As I've seen on the enginetorque-text-element and your fuel consumption dashboard that your app is able to display this data with less delay. Maybe you're using a filter to improve the display of the scale pointer.? I see that could be an issue. But maybe you can take into consideration to use the not delayed (not filtered) data for the numbers in the clocks, even if the scale pointer is a little slower. 
(By the way VW has the same issue in the AID. Consumption clock on the left is way slower than the numbers in the centre of the AID (MFA))

This app must have been a lot of work. Thank you for passion.

jilleb, 2021-08-08:
The Speedview library I'm using to display the clocks and values doesn't allow a more accurate view than this.

EA888DLBA, 2021-08-09:
> 
> 
> The Speedview library I'm using to display the clocks and values doesn't allow a more accurate view than this.

Thank you for your quick response and for the explanation. 
I just hoped that the numbers in the clock's center could use the same code as the 4 text elements. But if the numbers in the clocks are using the Speedview library as well, I see your point.
However request #130 can improve the situation for my use-case as well. So I think this request could be closed now, isn't it?