jilleb/mqb-pm #135: [BUG] Am I doing something wrong?

SpursMadDave, 2021-08-31:
Hi I have downloaded AAAD and it works perfectly with AA Mirror Plus allowing me to see Torque on my mib2.5 in my Golf R.

I thought I would purchase and try Performance Monitor but I just can't get the dials working. I have downloaded the latest version 1.4 and also dowoaded the VAG extention. In the app it shows VAG Extentions as telemetry provider.

I am using an Aawireless box but tried with a cable connection also and just can't seem to get the dials working, am I missing something obvious?

Thanks in advance for any help

jilleb, 2021-08-31:
From the readme:

>No data If you don't see any data in the app, it can have various reasons: Some cars don't provide any data to the headunit. Workaround: Use Torque with an OBD2 dongle to get data. Some MIB2 firmware versions don't provide data to the exlap channel. Not all data elements are available on all cars. If one element doesn't provide data, try an other one. Most cars do not report oil and coolant temperatures below 50. Most cars do not report any turbo pressure. Consider yourself lucky when your car does.

SpursMadDave, 2021-08-31:
Thanks for the quick reply, yes I am using an Elm BT adapter and my phone is getting data from it just not the mib, I will try some other dials and see if anything appears thanks

SpursMadDave, 2021-09-01:
Managed to get the clocks working with torque data but won't connect to VW (exlap) so maybe my unit mib2.5 is not compatible.
Look forward to any updates especially if they allow custom pids 👍