jilleb/mqb-pm #137: [BUG]

alex-christou, 2021-09-24:
***Description***
As shown in attached picture , app is visible in AA and not showing anything. Just the settings screen with a message that there are no new messages.
Notification sounds for RPM, steering angle and oil temperature are working.
Logs are collecting all kinds of data (attatching recent file as well)

App installed via AAAD following guides. (un/reinstalled a couple of times with the same output, permissions granted etc etc)

Skoda Fabia MK3 
Hardware: 696
Software: 8742
AA(car version) S1.3_A8742
Phone: RealMe7 pro
Android 11
AA version: 6.8.613354

***Screenshots***
[image]

***Attachments***
[car-20210921200938.log](https://github.com/jilleb/mqb-pm/files/7225579/car-20210921200938.log)

nuwaus, 2021-10-06:
I have the same issue, I'm trying to get Torque Pro data on Performance Monitor.

nuwaus, 2021-10-08:
> @nuwaus Nope. To my understanding it never shows the correct landing page when started in Android Auto. I cant tell if the problem is the app, the car, or the phone. Will give it a try with another device probably. I really wanted this to work.

I'm not sure it is not working because my phone isn't rooted. In that instance, I have an old nexus phone and I might attempt with it, with root permissions.

alex-christou, 2021-11-05:
UPDATE: This is not an app issue. It looks like it is not installed correctly on android auto. Removing the issue

closed 2021-11-05 by alex-christou

ale99viv, 2022-02-06:
I had the same problem and i solve this out follow this:
- go to phone settings;
- App setting, search Perofomance Monitor;
- enable all permission and disable enerrgy saving system.

I have a Xiaomi and this work for me