jilleb/mqb-pm #152: [BUG] Performance Monitor crashing on Skoda Fabia Monte Carlo 2017 infotainment system

BRND04, 2022-01-18:
I have the latest version of Android Auto installed, I have the latest operating system on my Samsung Galaxy S20+ (Android 12 (ONE UI 4.0)). 

I have the VAG 2.4 apk installed as well as Performance monitor...

I can open the performance monitor app on my phone and customize the settings, however, whenever I plug my phone into my infotainment system and select android auto and then select the performance monitor app... the screen goes black on  my infotainment system and I get a pop-up on my phone saying "performance monitor has stopped".

Why is this occurring? This is very annoying and quite frustrating. 

NOTE: I am using AAAD. I have developer options unlocked and ticked unknown sources and enabled developer app mode in Android Auto settings. Also my phone is NOT rooted by I can use performance monitor without root? yes?

UbayGD, 2022-01-28:
Something else strange happens to me in my Fabia. In Android Auto I only get the "Performance Monitor Settings" app. If I select that app I only get a black screen with a message saying that there are no records or similar.

ale99viv, 2022-01-30:
did you find a solution?