jilleb/mqb-pm #154: [BUG] Performance monitor app does not appear in Android auto

UbayGD, 2022-01-28:
Hello!
I am having trouble getting the app to add correctly to Android Auto but have not been successful. My car is a 2020 Skoda Fabia.

I think I have followed the steps correctly:
1 - Enable developer mode in Android Auto and unknown sources.
2 - Install VAG Extensions for Android Auto (from AAAD)
3 - Install Performance Monitor for AA (from AAAD)

By following these steps the only thing I have managed to do is to display a "Performance Monitor Settings" icon. Just in case, I also rooted the phone and installed an application to enable third-party apps on Android Auto, without success.
I don't know what else I can try. I don't think I'm missing anything.

Here I leave information about the AA app version and firmware in my car.

Android Auto: 7.1
Car software: 8742

I'm sorry that the images are in Spanish but I think you can understand them.
[image]
[image]
[image]

Greetings and thanks! :)

ale99viv, 2022-01-30:
did you find a solution?

Maverick-cmd, 2022-02-18:
This is for older versions of Android Auto, before version 3, current versions are version 7.x.xxxxxx-release. Also versions for other things are way out of date as well. It will not work unless you downgrade or reverse engineer and rewrite some code.

ale99viv, 2022-02-18:
I currently use Android Auto version 7.x.xxxxxx and since today Performance Auto start working perfectly on my skoda fabia 2019. I don't know the reason but it works. I have installed Performance Monitor 0.88.0 through AAStore

Maverick-cmd, 2022-02-18:
Interesting, my attempt was way more difficult. Everything works for you though?

ale99viv, 2022-02-18:
Yes everything works fine. I started seeing the indicators but they didn't work and after about 2 weeks they started working by themselves